Ultimaker/Cura #9739: Cura is extremly slow after the last upgrade

andreixk, 2021-05-01:
### Application Version

4.9.0

### Platform

Mac OS X

### Printer

FLSUN QQ-S

### Reproduction steps

1. Update Cura
2. Open it.

### Actual results

Opening the app takes about 20s (before the update, it took 3-5s).
Panning/viewing the model is very laggy/choppy.

### Expected results

Should work as before.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6409390/cura.log)

I tried adding  the project file, the window shows error that you do not support them.

fvrmr, 2021-05-03:
Hi @andreixk thank you for your report.
What version of MacOS are you using?

andreixk, 2021-05-03:
> Hi @andreixk thank you for your report.
> What version of MacOS are you using?

Catalina: 10.15.7

nallath, 2021-05-03:
Do you have the sidebar plugin enabled?

fieldOfView, 2021-05-03:
The cura.log that was included suggests SidebarGUI is not involved.

andreixk, 2021-05-03:
> Do you have the sidebar plugin enabled?

I don't know, but I didn't install or enable any plugins since starting to use the app.

nallath, 2021-05-04:
The logs indicate that booting cura took ~7 seconds. I'm also not seeing anything in the logs that would indicate any slowdown.

andreixk, 2021-05-04:
> The logs indicate that booting cura took ~7 seconds. I'm also not seeing anything in the logs that would indicate any slowdown.

i'm not sure how long is the "boot" technically, but i count it from the time i click the app icon to the time i can start using it.

nallath, 2021-05-04:
The boot time is all the loading right before the UI is created. This is generally the slowest bit, but in your case it seems like it's the UI generation that is much slower.

Ghostkeeper, 2021-05-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/6589.

closed 2021-05-13 by Ghostkeeper