Ultimaker/Cura #9740: Option to manually change layer height after reaching a specified height

BloodyRain2k, 2021-05-01:
### Is your feature request related to a problem?

I would really like the option to manually set the layer height after a certain height has been reached.

### Describe the solution you'd like

I'm currently working on a part that has a small but still inconveniently placed overhang, and the "Adaptive Layer Height" settings are sadly doing everything, except helping with this.

It would be very easy to save on prototyping time if I could manually tell the slicer where to step the layer height up and where a finer height would be better.

UI wise, it might work like "selecting" the affected region with the slider on the right, and then entering the new layer height in a field, like "Override Local Layer Height".
The overridden layer height should be saved for the approxiated height region of the model, so that you won't have to re-do it over and over after every slice, and an altered layer range should have a different color on the layer slider, like darker for thinner and brighter for taller?

### Describe alternatives you've considered

Printing everything in a thin layer: wasted time if only few sections need that.

Adaptive Layer Height: unintuitive and seems to only want to do the opposite of what I need, adding detail where I don't want it, and not adding it where I want it.

### Affected users and/or printers

People who prototype parts, especially ones that have longer undetailed sections that can be printed in thick layers.

fvrmr, 2021-05-04:
Hi @BloodyRain2k thank you for your feature request. 
I will discuss this with the team. Keep you posted.

Ghostkeeper, 2021-05-05:
This is a request for making a setting adjustable per layer. That is generally a pretty big feature to implement. The value for it is also wasted on everyone except the expert users, similar to ChangeAtZ, since it requires careful inspection of the layer data and then knowing what needs to be changed where.

Ideally you'd have some interface with segments of the layer range that can get different settings similar to the per-object settings, like how Simplify3D allows setting different processes for parts of the range of layers.

There are a few major hurdles to implement this though:

* We'd need to design an interface that is usable. Our layer view slider is rather small, but it needs to be precise enough and discoverable. Those requirements are at odds.
* We need to decide on some resolution for when the user changes settings at different levels. Do the per-layer settings override per-extruder settings? Do they override per-object settings?
* In CuraEngine we need to know how many layers constitute a certain vertical space. For instance, with a Support Z Distance of 0.3mm, and a layer thickness of 0.1mm, CuraEngine will determine that the support needs to start 3 layers below the model. If the layer thickness suddenly changes from one layer to the next, this distance will be wrong. The same goes for any other Z dimension, like top/bottom thickness, support interface thickness, etc. This is already a problem for Adaptive Layers too.

Because of these hurdles and the expertise needed to use such a feature, we've decided to defer this. It's currently not worth the effort to develop it and won't get priority. There are more valuable things to build.

yet-another-average-joe, 2021-05-20:
@BloodyRain2k : oh yes, it would be great. Didn't think of the UI, as explained by @Ghostkeeper .
Recently, I wish I had such a feature. I thought it could be achieved by support blockers and per model settings, not thinking of the UI issues. Now, I understand why layer height is not in the per model parameters list.

(a possible workaround)

Then I was thinking of slicing the same model with different layer heights, doing some copy/paste between the gcode files, and creating another one mixing layer heights... Heights are easy to search/find with Notepad++. A problem : what if the first one is sliced at 0.24mm and the other one at 0.1mm ? In order to merge the pieces of gcode, we should have to live with inaccuracies, and/or search for the smallest common multiple  (or use better layer heights than 0.1 and 0.24 !)

I finally gave up. But I'm sure it can be done (with some caveats). Coding a CLI app for parsing / merging two or more Cura gcode files, given a few heights or layer # and tolerances, shouldn't be too difficult... Maybe someone already did something similar somewhere on the web.