Ultimaker/Cura #9742: Slice function not working after update to Cura 4.9.0

DevilAle, 2021-05-01:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Anycubic Mega Zero

### Reproduction steps

1 Upgrade to Cura 4.9.0
2 Open .stl file
3 Slice it
4 Slice function not working
//*Even after restarting the PC or reinstalling Cura the Slice function still not working.

### Actual results

Slice function is not loading at all the % bar does not not even appear.

### Expected results

Slice working as before.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

fvrmr, 2021-05-03:
Hi @DevilAle thank you for your report. 
Could you provide me your log file? You can find that here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
Thank you in advance.

DevilAle, 2021-05-04:
Hi fvrmr, thank you for your reply.
Unfortunately I could not find the log-file on the path you provided me, specifically I do not have any AppData folder in C:\User\My_UserName.
I will try once again to install it with administrator right..

I will let you know.

Best regards

DevilAle, 2021-05-04:
Hi,
Short update, after reinstalling (3rd times) this time with administrator right and deleting previous configuration looks everithings working fine.

Thank you for your support.

closed 2021-05-05 by nallath