Ultimaker/Cura #9743: Insta-crash whenever the "Scale" button is pressed in Preview Mode

Enrico-Fanelli-97, 2021-05-01:
### Application Version

4.9.0

### Platform

Zorin OS 15.3

### Printer

Creality Ender-3 Pro

### Reproduction steps

I'm running Cura directly from the AppImage downloaded on the official site on my Linux machine. 

1. Pressed the "Scale" button while in Preview mode

This happens with every STL file.

### Actual results

Cura crashes.

### Expected results

Not crashing.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Cannot seem to find any log generated by Cura, instead here's the terminal output. After pressing the "Scale button" the last line is "Illegal instruction". [Log.txt](https://github.com/Ultimaker/Cura/files/6409696/Log.txt)

fieldOfView, 2021-05-01:
Duplicate of #9709, which incidentally also reports using Zorin OS

fvrmr, 2021-05-03:
Since it is a duplicate I will close this issue.

closed 2021-05-03 by fvrmr