Ultimaker/Cura #9744: Error setup

OdinochkA15, 2021-05-01:
### Application Version

4.9.0

### Platform

Windows 7

### Printer

Creality Ender-5

### Reproduction steps

I downloaded the new version 4.9.0 and started installing it. But an error occurs: The file cannot be opened for writing. Now the program is not removed from the Program files and is not installed normally. Put back version 4.8

### Actual results

Does not work

### Expected results

not work

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

[cura_log.zip](https://github.com/Ultimaker/Cura/files/6409748/cura_log.zip)

fvrmr, 2021-05-03:
Hi @OdinochkA15 thank you for your bug report. 
This might be related to this issue: https://github.com/Ultimaker/Cura/issues/9663
Can you check if that works for you?

OdinochkA15, 2021-05-03:
oddly enough, this fixed the installation error.  But the program itself does not start after installation.  cura.exe error

fvrmr, 2021-05-04:
What is the error you get when trying to open Cura?

OdinochkA15, 2021-05-04:
the second screen that I attached in the first message.  cura.exe error

OdinochkA15, 2021-05-04:
while version 4.8 works fine

closed 2024-12-09 by github-actions[bot]