Ultimaker/Cura #9745: skin overlap missing

pi4head, 2021-05-01:
### Application Version

arachne

### Platform

windows 10

### Printer

creality ender 3

### Reproduction steps

I activate skin overlap in the setting menu

[image]

### Actual results

it doesn't show up

[image]

### Expected results

it should show up.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

mandatory text here

fvrmr, 2021-05-03:
Hi @pi4head thank you for your report.
I am not able to reproduce this. Could you share a project file (file - save project)?

pi4head, 2021-05-03:
Here you go.
on a side note, on some preloaded profile that is imported from 4.8
version, the skin overlap is already there.

fvrmr, 2021-05-04:
Could you attach the file as a .zip in a comment on Github? I can't see the file since you replied via email.

pi4head, 2021-05-04:
[bugReport_arachne.zip](https://github.com/Ultimaker/Cura/files/6420062/bugReport_arachne.zip)
gladly.

fvrmr, 2021-05-04:
Thank you, if you change your top/bottom pattern to lines or zigzag the skin overlap is visible. The reason is that concentric is not supported with skin overlap. 
I will close this issue, since changing the top/bottom pattern will solve this issue.

But feel free to reach out if you have more questions.

pi4head, 2021-05-04:
thank you for your clarification. I think that an additional line in the tooltip would be helpful. Have a nice day!

closed 2021-05-04 by fvrmr