Ultimaker/Cura #9746: Adaptive layer height but limited to using multiples of magic number

nushipwright, 2021-05-01:
### Is your feature request related to a problem?

Adaptive layer height is a fantastic feature as it can effectively reduce print time while keeping the quality of prints fairly good. In theory this is fantastic.
However, often times adaptive layer height also means microstepping of the Z axis motors and on some low/mid end printers this results in uneven extrusions and ugly visible layers. Therefore compromising quality. I believe for some reason there is a mismatch between actual layer height and extrusion. Drivers unable to hold Z at a microstrip height?!

### Describe the solution you'd like

I would like to see an option to limit or constrain the adaptive layer heights (automatically chosen by the cura engine) to an X multiple of the "magic number". 
Always one full step height of the Z axis motor(s). 

Avoiding microstepping I believe could have significant benefits in the quality of prints.

### Describe alternatives you've considered

I am not a developer therefore I don't know how difficult or perhaps easy this feature would be to implement. 
One could (calculate and) define the value of one full step height in the slicer config and the adaptive layer height would be always an X multiple of that pre-defined height value.
Layer heights could still vary but would always stick to full steps on the motors.

### Affected users and/or printers

Most low/mid end printers.

fvrmr, 2021-05-04:
Hi @nushipwright thank you for your feature request.
I will discuss this with the team. Keep you posted.

Ghostkeeper, 2021-05-05:
Cura already knows the step size of the printer, at least if the printer definition file is complete. It is known in the [Steps Per Millimeter (Z) setting](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_steps_per_mm_z.md). The Adaptive Layers feature could be adjusted so that it snaps to those step sizes, assuming Z=0 is on a step interval.

However with the current status of Adaptive Layers in experimental, we've decided to defer this. There are much more pressing issues, such as that all Z distances and thicknesses are wrong, that would have to be fixed first.

d-schmidt, 2021-05-11:
Cura allows to define the apative layer step size. Cura might skip steps (which should be a bug) but you can set your magic number there,

nushipwright, 2021-05-12:
Thank you both for your replies!
Would you be so kind to tell me where I can define the adaptive step size
in cura?

Thank you!
Norbert

d-schmidt, 2021-05-12:
[image]

nushipwright, 2021-05-12:
Awesome, thanks!

Ghostkeeper, 2021-05-17:
The step size merely limits the speed of transitions in layer thickness. It doesn't limit the layer thickness to be a multiple of the step size. For instance, if you have a part of the print that should be printed at 0.1mm and another part that should be printed at 0.2mm layer thickness, with 0.04mm step size you'll still get the layers at thicknesses:

* 0.1mm
* 0.14mm
* 0.18mm
* 0.2mm.

So the transition from 0.1mm to 0.2mm is more gradual, but the last increment was 0.02mm, which might not align with your Z stepper motor's steps. In fact, if Z=0 aligned, then Z=0.1mm was the one that didn't align.

d-schmidt, 2021-05-17:
But this is caused by the max variation or start height not being a multiple of your step size. If you want constant 0.04 step size, you woud need to set max variation to 0.12 resulting in:  
0.1  
0.14  
0.18  
0.22  

you can't step correctly from 0.1 to 0.2 if your stepper works in 0.04 steps so you need to change one of these values. This is not a problem of adaptive layer height.

nushipwright, 2021-05-18:
Thank you for the clarification.

Ghostkeeper, 2021-05-19:
It's not just the result of the Maximum Variation. Maximum Variation determines the maximum and minimum, but the layer height doesn't always need to switch from maximum to minimum.

The optimal layer height that Adaptive Layers arrives at is the layer height that ensures that the borders of two layers don't exceed the [Adaptive Layers Topography Size](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/adaptive_layer_height_threshold.md) setting. So if that is set to e.g. 0.5mm it will reduce the layer thickness such that the greatest difference between two adjacent layers is 0.5mm. And then that is clipped to the Maximum Variation setting. But this thickness can become anywhere between the minimum and maximum, even a layer thickness that is not a multiple of the step size.

So I'd argue that Cura currently doesn't really provide a solution for this, not even a workaround. You can kind of steer it a bit, but usually there will be many layers not on step size increments.

d-schmidt, 2021-05-19:
This isn't how adaptive seems to work. Either the layer height preview is lying or you are wrong.

Layer height: 0.2
Max Variation: 0.08
Step Size: 0.04

slices a the model using:
0.12  
0.16  
0.20  
0.24  
0.28

The Topography just controls how low it goes and at which overhang angle but it keeps to step size and variation. The layer height doesn't show weird smaller steps.

Ghostkeeper, 2021-05-19:
I looked into the source code to verify, and it turns out you're right, and I had the wrong idea about how this works!

https://github.com/Ultimaker/CuraEngine/blob/c6cd0f649157ec7bd4dc2e5b4a8697d4c0719814/src/settings/AdaptiveLayerHeights.cpp#L45-L58

I'll probably need to adjust this in the Settings Guide too. The Adaptive Layers articles there need some love anyway with more graphics.

So basically then we're concluding this is already implemented? The algorithm I linked to explicitly gives a set of "allowed layer heights" based on the variation step size, which is pretty much exactly what this feature request asks for. Let me know if you think this is not a sufficient solution.

closed 2021-05-19 by Ghostkeeper