Ultimaker/Cura #9747: M1 preview glitches

DumbGameMaker, 2021-05-01:
### Application Version

4.9.0

### Platform

macOS Big Sur 11.2

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. get some stls to slice - put a bunch of large ones, or some that are not manifold.
2. slice
3. preview
4. oh no, a bunch of stuff happens

### Actual results

can't really explain it, so refer to these images:
[image]
[image]

it normally looks way worse, of course when I try to screenshot it, it gets less bad. also, it sometimes will crash.

log file below:

[cura.log](https://github.com/Ultimaker/Cura/files/6410509/cura.log)

### Expected results

we should see the preview.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Looking at the log file, it could be a QT issue. Cura uses an older version of QT which doesn't yet fully support macOS Big Sur/M1. Updating would require dropping support for a lot of macOS versions.

DumbGameMaker, 2021-05-03:
ok i tried it again and it seems to have been fixed. i realized that i wasn't actually on 4.9.0, i was on 4.8.0. it is fixed on 4.9.0. i should have known cuz 4.9.0 has a red outline in preview while 4.8.0 doesn't

DumbGameMaker, 2021-05-03:
ok nevermind, on 4.9.0 with 100% infill it works fine. but you wouldnt use 100% infill - i switch to 40% infill and my computer freezes. I can't open force quit, and the power button does nothing. I am forced to UNPLUG my computer. plsfixthx

Ghostkeeper, 2021-05-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/9107 .

The Qt platform doesn't support Apple M1 yet, so neither does Cura which is based on Qt.

closed 2021-05-17 by Ghostkeeper