Ultimaker/Cura #9748: Clicking the download button for Cura 4.9 does not start the download

Geneva1560, 2021-05-02:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

The button to initiate the download for Cura 4.9 is not working. I tried it in Google Chrome, Modzilla Firefox, Microsoft Edge, Internet Explorer, and the results were all the same. I would like to use Cura 4.9 because I needed to use it to print a file that I had but since I am not able to download it, I guess I will never know.

### Actual results

I give up and did something else. There is nothing that I can do. Clicking the download button does not start the download at all for Cura 4.9.

### Expected results

Cura 4.9 should start to download and then I could set it up, but the download is not starting when I click the button.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura 4.8 was not having this issue. I downloaded Cura 4.8 very recently and the download started when I clicked the button but not for Cura 4.9. I am not sure why.

fieldOfView, 2021-05-02:
Which button, where?

You can download Cura 4.9 here: https://github.com/Ultimaker/Cura/releases/tag/4.9.0

Geneva1560, 2021-05-02:
> Which button, where?
> 
> You can download Cura 4.9 here: https://github.com/Ultimaker/Cura/releases/tag/4.9.0

it's on their website. It's this link. You click on 'Download for free' and choose what operating system you have. None of the buttons actually do anything and none of them start any kind of download when you click on them.

https://ultimaker.com/software/ultimaker-cura

Thank you also for supplying me with that other link, I've been able to download it now. I didn't know that I could downloads it another way.

Geneva1560, 2021-05-02:
I found out what I did wrong. I didn't scroll down enough to see the blue box that says download. I accidently used the mouse wheel and the download button was at the bottom. I don't remember seeing that before but there it is. Problem solved I guess.

closed 2021-05-02 by Geneva1560