Ultimaker/Cura #9749: print "one model at a time"

giuliosaoncella, 2021-05-02:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Artillery Genius

### Reproduction steps

Tryied to print multiple models using "one at a time" function (translated from italian, I don't know the exact function name in English version...)

### Actual results

The printing process "takes a breath" between every model: this causes PLA filament molding down from the extruder, causing garbage and even dameges to the model

### Expected results

please consider adding a "pause between multiple printing ON / OFF" function... if someone likes things as they are right now or -like me- prefers a "continuos" printing, one model at a time... thanks

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

...no additional infos...

fvrmr, 2021-05-03:
Hi @giuliosaoncella thank you for your report. 
I would see this more as a feature request. 
But if you want to add a pause yourself you could use the post processing script search and replace. 
And search for `;LAYER:0` and replace it with: `;LAYER:0\nM0` as an example. 
Let me know if this would work for you.

FPrefect20, 2021-05-04:
Hi @giuliosaoncella!
can you pls. check whether you are using different temperature settings for different layers.
Meaning for example 220°C for first layer / 200°C for the rest of the object.

In that case CURA will wait for the temperature to rise to the first-layer-temperature-setting before printing the second object.
This would cause to a "waiting breath" like you described.

Ghostkeeper, 2021-05-17:
Similarly, also make sure that the initial layer build plate temperature is the same as the normal build plate temperature.

closed 2021-05-17 by no-response[bot]