Ultimaker/Cura #9752: PauseAtHight script wrong extract in Repetier(@Pause) mode

Gleis1234, 2021-05-03:
### Application Version

4.9.2

### Platform

Kubuntu Linux

### Printer

Anycubic Chiron

### Reproduction steps

Slice with G-code modification PauseAtHight:
1 ;added code by post processing
2 ;script: PauseAtHeight.py
3 ;current layer: 10
4 M83 ; switch to relative E values for any needed retraction
5 G1 F6000 E470              ; Filament should be retracted not extracted
6 G1 F300 Z3.2 ; move up a millimeter to get out of the way
7 G1 F9000 X20 Y190
8 G1 F300 Z17.2
9 M84 E0
10 @pause now change filament and press ; Do the actual pause
11 G1 F6000 E470             ; Now Filament can be extracted
12 G1 F200 E125
13 @info wait for cleaning nozzle previous
14 @pause remove the waste filament parking area continue
15 G1 F6000 E-470            ; why retract here full length ?
16 G1 F300 Z2.2
17 G1 F9000 X290.299 Y71.113
18 G1 F6000 E470             ; and extract here again ?
19 G1 F2400 ; restore extrusion feedrate
20 M82

### Actual results

When arriving the pause state the filament is extracted by the full retraction length at speed 100 mm/sec (line 5) . At this time the "old" material is already in the nozzle and can not be extruded by full speed. The extruder rotates without moving the material, only abrasion happens. Here the material should be retracted, so it can me removed and a "new" material can be inserted. When continuing the next steps are correct. The filament is extracted (line 11) and with an extra amount the nozzle is cleaned.
Then the material is retracted again (line 15), the head moves to the object and the material is extracted (line 18) Why this retraction/extraction here?

### Expected results

When in the pause state the material should be retracted not extracted.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

- see code snippet above

fvrmr, 2021-05-03:
Hi @Gleis1234 thank you for your report.
Could you share a project file (file - save project)? Because I am not able to reproduce this, then I can take a closer look.

Gleis1234, 2021-05-03:
How can I send a *3mf file? Or do you need the gcode as a text file?

Gleis1234, 2021-05-03:
Another try with this address

Hi Fenne,

I try to send the project file in this email. I found no way how to send
it in github.

fvrmr, 2021-05-04:
Via email is also not working. 
You could share it as a .zip file in a comment on Github.

Gleis1234, 2021-05-04:
Here the compressed project file
[AC_Part73bmod.3mf.gz](https://github.com/Ultimaker/Cura/files/6423058/AC_Part73bmod.3mf.gz)
I modified the gcode by hand as follows:

1 ;added code by post processing
2 ;script: PauseAtHeight.py
3 ;current layer: 10
4 M83 ; switch to relative E values for any needed retraction
<strong>5 G1 F6000 E-470 ; Filament should be retracted not extracted</strong>
6 G1 F300 Z3.2 ; move up a millimeter to get out of the way
7 G1 F9000 X20 Y190
8 G1 F300 Z17.2
9 M84 E0
10 @Pause now change filament and press ; Do the actual pause
11 <strong>;G1 F6000 E470 ; Now Filament can be extracted</strong>
12 G1 F200 E125
13 @info wait for cleaning nozzle previous
14 @Pause remove the waste filament parking area continue
15 <strong>;G1 F6000 E-470 ; why retract here full length ?</strong>
16 G1 F300 Z2.2
17 G1 F9000 X290.299 Y71.113
18 G1 F6000 E470 ; and extract here again ?
19 G1 F2400 ; restore extrusion feedrate
20 M82

Ghostkeeper, 2021-05-17:
It's actually a bit weird to me that the PauseAtHeight script instructs you to change filament. That's what the FilamentChange script is for.

But yes, I do see these mistakes and could reproduce them. Perhaps this has gone wrong when the multitude of pause scripts were merged into one.

Gleis1234, 2021-05-17:
The FilamentChange script was my first try, but the M600 G_Code is not supported by Anycubic firmware. Therefor I found the @pause solution a good way to achieve my goal. If you can implement the @pause in the FilamentChange script, this would also do it.

commit 4d4427ce (2021-05-18): Extrusion should be retraction.

Should fix #9752 aka internal ticket CURA-8246 -- This makes the repetier case completely analogous to the other 'not-griffin' flavours (at least what concerns retraction amounts), and also neatly dove-tails with the first point things start to go wrong in the bug-report.
PR #9864 (2021-05-18): [CURA-8246] Pause-at-height/repetier: Extrusion should be retraction.
Should fix #9752 -- This makes the repetier case completely analogous to the other 'not-griffin' flavours (at least what concerns retraction amounts), and also neatly dove-tails with the first point things start to go wrong in the bug-report.
closed 2021-05-18 by Ghostkeeper