Ultimaker/Cura #9753: Z-Seam Always at Outside Wall of Object

FPrefect20, 2021-05-03:
### Application Version

4.9.0

### Platform

Win 10

### Printer

Anycubic Mega S

### Reproduction steps

Build a simple box with open top.
Slice with cure 4.9 using settings:
- z-seam alignmeng "user defined"
- coordinates representing the inside of the object

### Actual results

Z-Seam visible in cure (white line) and at the outside of the printed object
(no matter what coordinates are entered)

### Expected results

Z-Seam at the inside (surface of the inner wall)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

fvrmr, 2021-05-03:
Hi @FPrefect20 thank you for your report. The white line also represents the starting point. 
But could you share your project file (file - save project) so I can take a look at your model and settings.

FPrefect20, 2021-05-03:
If [-105mm] / [+105mm] is representing the [leftmost / rightmost] "corner" of the object, [0,0] should be the center, right?

[image]

FPrefect20, 2021-05-03:
(pls. see project file attached)

[TEST_BOX.zip](https://github.com/Ultimaker/Cura/files/6416511/TEST_BOX.zip)

fvrmr, 2021-05-04:
Hi @FPrefect20 thank you for your project file and screenshot. Now I understand want you would like to have.
You can't get the seam in the middle of the box because a seam will only start at a corner. 
You can find more information about the z seam here: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/z_seam_type.md

Let me know if you have more question. But for now I will close this issue.

closed 2021-05-04 by fvrmr

FPrefect20, 2021-05-04:
Hi @fvrmr !
yes, i have more questions - thanks for the offer to ask.  :-) 
Why does it need a corner for a seam? why not place it on a flat wall for example? (there are other slicers doing so)
And if i slice a circle it doesn´t need a corner for the seam to be placed at.

fvrmr, 2021-05-06:
The seam will always start at a corner, doesn't matter how big the corner is. So a circle has just tiny sides. And on the line where the sides connect the z seam can be added. 

See here: 
[image]

And this is how the seam will look in Cura with that same model:
[image]

Hope this explains how the seam works.

Ghostkeeper, 2021-05-17:
For more of an explanation on why seams in Cura always start in a corner, see https://github.com/Ultimaker/Cura/issues/8010 .

It's a minor limitation of the data structures in CuraEngine. It simply tracks an integer at which vertex it should start printing a loop, so the seam is then always on a vertex.