Ultimaker/Cura #9756: 4.9 won’t load

jameslunt81, 2021-05-03:
### Application Version

4.9.0

### Platform

Windows 7

### Printer

Creality cr10s pro

### Reproduction steps

Upgrade from 4.8
4.9 fails to load, goes through initial screen initialisation of plug ins then closes program.
Occurs every attempt to load.

### Actual results

Program won’t load

### Expected results

Program should load

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Won’t load

fvrmr, 2021-05-03:
Hi @jameslunt81 thank you for your bug report.
Does the installation fails or when you start up 4.9? If it is a start up, could you share your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Goodfeat, 2021-05-03:
[image]

I also stopped loading cura 4.9. reinstallation with saving configuration did not help

Goodfeat, 2021-05-03:
In general, I think this is a flaw in the program.

As it turned out, the program was trying to access a network drive (the text directly indicates this error). but I have no idea why the next time the program starts, it persistently checks the network drive. I think this point should be "added to the exception"

it turns out that this error does not apply to the topic

fvrmr, 2021-05-04:
Hi @Goodfeat your bug is related to this issue: https://github.com/Ultimaker/Cura/issues/9647
Currently we are working on a fix for that. In the mean time you could try to rename your X drive.

TiRaNy4u, 2021-05-06:
> Hi @Goodfeat your bug is related to this issue: #9647
> Currently we are working on a fix for that. In the mean time you could try to rename your X drive.

In my case is renaming the X-drive not possible. X-drive used in other things. Yeah, some people do use the X as a drive letter :)

nallath, 2021-05-06:
> In my case is renaming the X-drive not possible. X-drive used in other things. Yeah, some people do use the X as a drive letter :)

We have a confirmed fix for this. We plan to release a 4.9.1 that includes this.

Goodfeat, 2021-05-06:
In my case I was also unable to rename the drive. I had to remove the disk via the cmd

TiRaNy4u, 2021-05-06:
> > In my case is renaming the X-drive not possible. X-drive used in other things. Yeah, some people do use the X as a drive letter :)
> 
> We have a confirmed fix for this. We plan to release a 4.9.1 that includes this.

Any ETA on the 4.9.1?

nallath, 2021-05-06:
> > > In my case is renaming the X-drive not possible. X-drive used in other things. Yeah, some people do use the X as a drive letter :)
> > 
> > 
> > We have a confirmed fix for this. We plan to release a 4.9.1 that includes this.
> 
> Any ETA on the 4.9.1?

Don't have a final date yet, but I expect it to be released next week.

closed 2021-05-17 by no-response[bot]