Ultimaker/Cura #9758: "Modify Overlaps" dialog disappears as soon as "Select Settings" is clicked

KenLucke, 2021-05-03:
### Application Version

4.9.0

### Platform

Mac OS 10.15.7 (Catalina)

### Printer

N/A

### Reproduction steps

1. Select a Support Blocker
2. Click "Per Model Settings"
3. Click "Modify Settings for Overlap"
4. Change from "Infill Mesh" to "Cutting Mesh" or vice-versa
5. Click "Select Settings"

::poof:: Per Model Settings dialog disappears.   

https://user-images.githubusercontent.com/57936294/116895897-80681800-abe8-11eb-91d4-17046f96b967.mp4

### Actual results

HOWEVER, if you select the support blocker (or other model) in the Object List, everything works as it should.  The FIRST time.   But if you change from Infill to Cutting or vice versa again, the dialog disappears when you click "Select Settings" until you reselect.

https://user-images.githubusercontent.com/57936294/116896363-06845e80-abe9-11eb-9e93-06d04ca625f3.mp4

### Expected results

The normal Select Settings dialog should appear and be accessible.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Uploading Butterfield Statue model.3mf.zip…]()

fvrmr, 2021-05-04:
Hi @KenLucke thank you for your bug report. 
I will discuss this with the team. Keep you posted.

Ghostkeeper, 2021-05-05:
We've been able to reproduce this issue on another MacOS, but not on any other computer. So we're going to assume it is MacOS specific. We've added it to our backlog.

Developers, see issue CURA-8217.

KenLucke, 2021-05-05:
Just a data point:  I just tried this on Windows 10, and the same thing happened.... BUT, my Windows 10 runs in my instance of Parallels Desktop on my Mac.  Theoretically, it's emulating on the bare hardware, so being on a Mac should not influence it, but I'm not versed enough in the guts of things any more to know for sure.

Ghostkeeper, 2021-06-14:
I have a question for those who are able to reproduce this bug. Does it also occur if you execute these steps?

1. Select any model.
2. In the per-object settings tab, add any setting (e.g. Infill Pattern).
3. Click on the minus symbol to remove the setting again.

Does the panel close at step 3?

Ghostkeeper, 2021-06-15:
I've received word from a colleague that the panel does not close, so my initial suspicion was incorrect.

However I had a new hunch and implemented a fix I'm pretty hopeful about: https://github.com/Ultimaker/Cura/commit/f18e5f96abf89a7b0f5797ff5514799056077224

closed 2024-12-09 by github-actions[bot]