Ultimaker/Cura #9760: Shifting part of the object

absolutesh, 2021-05-03:
### Application Version

4.8.0

### Platform

Linux

### Printer

Anycubic Mega Zero

### Reproduction steps

1. Added шахматы_UndeadQueen.stl
2. Made it 86.47% in all dimentions
3. Added шахматы_UndeadKing.stl, moved a bit closer
4. Sliced and run preview - all looks fine
5. Printed the gcode file and the bigger object was shifted at about 32.3 mm from the base[

Project file is https://github.com/absolutesh/gihubattachments/blob/master/AMZ_%D1%88%D0%B0%D1%85%D0%BC%D0%B0%D1%82%D1%8BUndeadQueen-z-King.3mf
Stls:
https://github.com/absolutesh/gihubattachments/blob/master/%D1%88%D0%B0%D1%85%D0%BC%D0%B0%D1%82%D1%8B_UndeadKing.stl.gz
https://github.com/absolutesh/gihubattachments/blob/master/%D1%88%D0%B0%D1%85%D0%BC%D0%B0%D1%82%D1%8B_UndeadQueen.stl.gz

Gcode output:
https://github.com/absolutesh/gihubattachments/blob/master/AMZ_%D1%88%D0%B0%D1%85%D0%BC%D0%B0%D1%82%D1%8BUndeadQueen.gcode.gz

### Actual results

Objects are printed as look on preview

### Expected results

The bigger object was shifted at about 32.3 mm from the base

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[AMZ_шахматыUndeadQueen.gcode.gz](https://github.com/Ultimaker/Cura/files/6417818/AMZ_.UndeadQueen.gcode.gz)
[шахматы_UndeadKing.stl.gz](https://github.com/Ultimaker/Cura/files/6417830/_UndeadKing.stl.gz)

[AMZ_шахматыUndeadQueen-z-King.3mf.zip](https://github.com/Ultimaker/Cura/files/6417859/AMZ_.UndeadQueen-z-King.3mf.zip)

fvrmr, 2021-05-04:
Hi @absolutesh thank you for your report. 
I can't see anything strange in the project files or gcode. So it must be an issue with your printer configuration.

Did you also try to slice your models with Cura 4.9?

absolutesh, 2021-05-04:
I believe, however, it is a slicer's fault as entire object was correctly wrapped into the tree-like support structure from bottom to the top. See the pictures of how it was printed.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

absolutesh, 2021-05-04:
I will not pursue this issue further as I just don't need more prints of this so won't waste any more filament on it. If I'll see any other problem I'll file a new bug then.

absolutesh, 2021-05-04:
Now I believe that's possible a result of the printing head or cooling fan hitting the base at height about 32.3 mm as there is an upper expansion zone and it happens so if that was not enough adhesion for the base and it shifted towards support surface with no further movement and as there was still support around it, it continued to be printed this way.

It opens to another question then: if filament expands towards printing head when printing elevating edges so it starts hitting printing head or cooling fan, is there any way to prevent it (as you can see I set Z hop at retraction to 0.8 mm so it doesn't hit it during switching between different printing zones so it seems it happens during printing the same zone, e.g. from outer wall to infill zone)?

nallath, 2021-05-05:
> It opens to another question then: if filament expands towards printing head when printing elevating edges so it starts hitting printing head or cooling fan, is there any way to prevent it (as you [...]

I've had a few prints fail because of this as well. Setting the Z-hop also doesn't really solve the issue. I've been able to save prints when i saw it happen by decreasing the speed of the print (so the nozzle has more time to push heat into the raised bit)

closed 2021-05-05 by nallath