Ultimaker/Cura #9761: Unnecessary Traveling Can Not be Removed

FPrefect20, 2021-05-03:
### Application Version

4.9.0

### Platform

Win 10

### Printer

Anycubic Mega S

### Reproduction steps

Slicing objects with setting "fill gaps between walls" is causing a complete printing turn with extra travel moves only for filling the gaps.
There is no possibility to combine "filling gaps" with "infill" or "walls" to save travel moves for better results.
To reproduce pls. see enclosed test-file.

### Actual results

CURA calculated printing process:
- print walls
- print infill (with travelling over the complete object)
- print "gaps between walls" (again travelling the complete object)

### Expected results

combine printing "gaps" with "infill" for saving travel moves, retractions and printing time.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

even with setting "optimize" the "gaps" are printed separately

[image]

Project file of sample:

[TEST_BOX02.zip](https://github.com/Ultimaker/Cura/files/6418306/TEST_BOX02.zip)

fvrmr, 2021-05-04:
Hi @FPrefect20 thank you for your report.
This is expected behaviour. Since the walls, gaps and infill are all different types. 

But I will discuss this issue with the team as a feature request to minimize travel moves while printing gaps.

Ghostkeeper, 2021-05-05:
This issue is fixed in the Arachne version of Cura, which is currently in beta: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta

There is no gap filling any more in that version of Cura. There is something that functions similarly though, because it can fill thin parts with an odd number of walls, not needing to loop every wall around. In cases of a thin corner like here, it functions similar, except that these odd walls are printed together with the rest of the walls, precisely what you ask.

We're currently still investigating the impact of this on the print quality. There has been some feedback that this reduces print quality if the walls are printed before the outer wall, since the walls can push the rest of the material further out, leading to shine-through. But we're not sure if that is really an issue yet.

We won't fix this for version 4.10 (which is still released without Arachne) but for Arachne it is fixed.

closed 2021-05-05 by Ghostkeeper