Ultimaker/Cura #9764: Disabling "Fill Gaps Between Walls" Produces Open Objects

FPrefect20, 2021-05-04:
### Application Version

4.9.0

### Platform

Win 10

### Printer

- any printer -

### Reproduction steps

Slicing object with setting "Fill Gaps Between Walls" = "Nowhere"

### Actual results

Top layer of the object is also affected by this setting.
The object is kept open at top layer

### Expected results

Expected is this setting only is affecting the gaps between the filling and the wall. 
Expected is that the object still will be printed with a closed top layer as given in "Top layer" setting.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2021-05-04:
In Cura use File | Save Project and post the 3mf file here.  You might have to Zip it.  It will have the model and your settings in the file.

FPrefect20, 2021-05-04:
Project File:
[TEST_BOX03.zip](https://github.com/Ultimaker/Cura/files/6421926/TEST_BOX03.zip)

GregValiant, 2021-05-04:
Well, it's the model...and it's not the model.  I sliced it with Cura 4.9, Arachne, IdeaMaker, and Prusaslicer and got the same results each time.  I adjusted the line width and there was no change.  Sent it to Netfabb for repair and there was no change.  Going to a .2 nozzle worked and gave me a clue.
#1 Your Initial Layer Line Width of 140% is causing problems with the first layer.  I would set that to 100%.
Because the walls are so thin, all the slicers are having trouble getting everything in there.  Just changing the Wall Line Count to 2 fixed the problem with the top and bottom skins and everything looks as it should.  I could not get infill into those little gaps.

nallath, 2021-05-05:
Alternatively, you could also try the Arachne Beta release. I expect that to do a much better job for objects like this.

FPrefect20, 2021-05-05:
@GregValiant : Thanks for your analysis!
although the corner has a multiple value of 0.4 mm (it is 2.0mm) the slicer can not get it filled (at top layer) with 5 lines.
changing the layer width to 0.39mm and it works fine for the top and bottom layer.

IMHO this is still a bug because the slicer is recognizing the gap (it is filled with option "fill gaps between walls")
Therefore - IMHO - the slicer should always fill small gaps at TOP and BOTTOM layer - no matter what setting is chosen for the space between INFILL and WALLS.

GregValiant, 2021-05-05:
It's certainly repeatable with that model and it was repeatable across multiple slicers.  I've had models where Cura put infill into narrower gaps than that.  I could tell because the extruder would skip as there wasn't enough room for the infill.  In the case of that model, it looks like the infill should have fit.
As to why the top and bottom skins were ignored, I have no idea BUT the other slicers did it too.  That's why I say "It's the model, but it's not the model".
Is it a bug?  It acts like a bug (there should be skins) and it's repeatable so I think it is.  One of the Cura Team will take a look and make a call.

Ghostkeeper, 2021-05-17:
> Expected is this setting only is affecting the gaps between the filling and the wall.

This is not what I'd expect. "Fill Gaps Between Walls" sounds an awful lot like it would affect gaps between two walls.

> Expected is that the object still will be printed with a closed top layer as given in "Top layer" setting.

This is also not the currently expected behaviour. The setting makes no distinction as to whether the walls are next to a piece of skin, a piece of infill or both. This area is considered part of the walls, not the skin or the infill, so there is no separation based on the skin thickness.

We could make that a feature request then. Not sure what should happen then with gaps in walls that are next to both skin and infill.
[image]

qwertymodo, 2021-05-18:
I would like to add my support to considering this feature request.  If you don't want to force new behavior, perhaps in the drop-down menu where the only options are currently "Everywhere" and "Nowhere" you could add a third option "Top and bottom layers only" which would then apply the fill pattern only to the top and bottom layers (tied to the top/bottom layer count setting)

Ghostkeeper, 2021-06-01:
What do you think should happen with gaps where half of the area is close to the top surface, and half is not? That is the case I illustrated in the above screenshot.

qwertymodo, 2021-06-01:
In a situation where it's near both skin and infill, the proximity to skin should be prioritized. The reason for filling the gaps is to produce better results, the reason not to is to print faster, and the idea of this feature request is to gain the benefits of speed while still getting the quality where it matters, so if it's ambiguous, best to err on the side of assuming it matters and fill there.