Ultimaker/Cura #9765: Sign in not working the first time, causing crash the second try (python not signed by codesign)

Schokobecher, 2021-05-04:
### Application Version

4.9.0

### Platform

macOS 11.3

### Printer

Artillery Sidewinder X1

### Reproduction steps

1. Start Cura
2. Click 'Sign in' in the upper-right corner
3. Click the 'Sign in' button
4. Get redirected in the default browser and sign in to Ultimaker Account
5. Get a success message in the web browser, but Cura still offers the sign-in option
6. Click again and get the Crash reporting dialog

### Actual results

Cura does not sign into the Ultimaker account despite the Web App claiming a successful authentication. Trying again without closing Cura yields a traceback which indicates python is not signed with codesign.

### Expected results

Sign in to the Ultimaker account first try

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Error traceback:
`Traceback (most recent call last):
  File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/site-packages/keyring/backends/macOS/__init__.py", line 49, in get_password
  File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/site-packages/keyring/backends/macOS/api.py", line 101, in find_generic_password
  File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/site-packages/keyring/backends/macOS/api.py", line 43, in raise_for_status
keyring.backends.macOS.api.SecAuthFailure: (-25293, 'Security Auth Failure: make sure python is signed with codesign util')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Toolbox/src/Toolbox.py", line 144, in _restart
    self._fetchPackageData()
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Toolbox/src/Toolbox.py", line 193, in _fetchPackageData
    self._makeRequestByType("packages")
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Toolbox/src/Toolbox.py", line 547, in _makeRequestByType
    self._application.getHttpRequestManager().get(url,
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/UM/TaskManagement/HttpRequestManager.py", line 128, in get
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/UM/TaskManagement/HttpRequestManager.py", line 273, in _createRequest
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/UM/TaskManagement/HttpRequestScope.py", line 57, in requestHook
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/cura/UltimakerCloud/UltimakerCloudScope.py", line 22, in requestHook
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/cura/API/Account.py", line 256, in accessToken
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/cura/OAuth2/AuthorizationService.py", line 135, in getAccessToken
  File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/cura/OAuth2/KeyringAttribute.py", line 36, in __get__
  File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/site-packages/keyring/core.py", line 55, in get_password
  File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/site-packages/keyring/backends/macOS/__init__.py", line 55, in get_password
keyring.errors.KeyringError: Can't get password from keychain: (-25293, 'Security Auth Failure: make sure python is signed with codesign util')
`

[cura.log.zip](https://github.com/Ultimaker/Cura/files/6422279/cura.log.zip)

It seems Cura is using its python version?  My default version is 3.9 (vs. 3.8 used by Cura)

 $ which python: `/usr/bin/python`
 $ which python3: `/usr/local/bin/python3`

I signed the version from homebrew (in /usr/local/bin/) manually but that did not resolve the issu

fvrmr, 2021-05-06:
Hi @Schokobecher thank you for your bug report.
Could you try following these steps:
1. go to Keychain Access app in the Utilities folder
2. under the File menu, choose Lock Keychain "Login"
3. go to the File menu again and choose Unlock Keychain "Login"

Please let me know if this fixes your log in problem.

Schokobecher, 2021-05-06:
Hi @fvrmr,

I checked that due to finding the same error on another project. My Keychains are unlocked and I have no option to lock them manually:

[image]

jphansendk, 2021-05-14:
I have the same issue  in 4.9.1 Big Sur

evtrados, 2021-05-26:
Hi @Schokobecher,
Some additional facts from my side. All Python related/dependent libraries are properly signed using codesign, otherwise the .app would not get notarized by Apple, which would then lead to it not opening at all on Big Sur, since it requires a successfully signed and notarized application bundle.
If you go to KeyChain Access, select the "All Items" category on the bottom left and then search for "cura" in the top right search field, are there any entries displayed? You should typically see 2 entries, one being the "refresh_token" and the other the "access_token" (you can see this after selecting the respective entry).
In general, from what I've read this seems to be totally system dependent, because none of us can reproduce the issue on Mojave, Catalina and Big Sur. Could you send us some more information about your system?

Zankar94, 2021-05-27:
Cura logged in for me when I  allowing all applications to access item under "access Control" after I searched Cura in Keychain app.

Zankar94, 2021-05-27:
[image]

Schokobecher, 2021-05-27:
Can confirm that the workaround by @Zankar94 works

@evtrados
Yes, both items are there (**refresh_token** and **access_token**). Maybe I have something in common with @Zankar94  and @jphansendk ?
What exact information do you need about my system/configuration?

jphansendk, 2021-05-27:
> Can confirm that the workaround by @Zankar94 works
> 
> 
> 
> @evtrados
> 
> Yes, both items are there (**refresh_token** and **access_token**). Maybe I have something in common with @Zankar94  and @jphansendk ?
> 
> What exact information do you need about my system/configuration?

Sorry haven't had time to test the workaround yet.. Will get back asap

jphansendk, 2021-06-17:
Thanks @Zankar94 Confirmed.. Better late than never ;-)

mpaw, 2021-07-21:
I have the same issue in Cura 4.10.0 on Big Sur. I tried manually adding Cura access for those two keychain items:

[image]

But it still doesn't log me in on first click and crashes on the second click. Error is same as what the author of this PR got.

Selecting `Allow all applications to access this item` on the keychain items **_does_** resolve the issue but seems to me like a bad idea. I don't want other applications to possibly grab my Ultimaker signed JWT.

Zankar94, 2021-07-21:
Couldn't you just select the confirm before allowing option and check the ask for keychain password box to avoid that?

mpaw, 2021-07-21:
@Zankar94 No, selecting `Confirm before allowing access` causes the app to crash with same error. Only `Allow all applications to access this item` logs me into Cura. Besides, even if it did work, it's not a solution to the underlying problem.

closed 2024-12-09 by github-actions[bot]