Ultimaker/Cura #9767: Ungrouped custom supports moved on reload of project file

KenLucke, 2021-05-04:
### Application Version

4.9.0

### Platform

Mac OS 10.15.7 (Catalina)

### Printer

N/A

### Reproduction steps

Using the attached project file:

1. Load file,  Note that the model and two custom support cylinders are a group, and import together, centered in the bed
2. Ungroup objects  Now there is a model and two support cylinders.
3. Resave project.
4. Reopen project just saved.
5. Support cylinders have moved 47mm+change to the right, main model remains centered.
6. Click two cylinders, set X location to zero, both return to the original position.
7. Repeat ad infinitum.

### Actual results

Ungrouped custom supports move upon loading of project file.

Top: open project with object/custom supports grouped
Bottom: opening same project saved with supports ungrouped from model.

[image]

[image]

### Expected results

They shouldn't move, even if ungrouped.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Butterfield Statue model3.3mf.zip](https://github.com/Ultimaker/Cura/files/6423597/Butterfield.Statue.model3.3mf.zip)

fvrmr, 2021-05-06:
Hi @KenLucke thank you for your bug report.
It looks similar to this issue: https://github.com/Ultimaker/Cura/issues/9287
But now I can reproduce it with my own models. I will discuss this again with the team. 
Keep you posted!

fvrmr, 2021-05-07:
I have discussed it with the team and created a ticket on our backlog.
Don't expect this to be a priority, but we do think that this should be fixed.

Developers see CURA-8226