Ultimaker/Cura #9768: Machine Settings Window does not refresh multi-line text boxes when resized.

thantik, 2021-05-04:
### Application Version

4.8+

### Platform

Windows 10

### Printer

Generic

### Reproduction steps

Resize the Machine Settings window large enough to expose more lines of text in the beginning and ending gcode boxes located at the bottom.

### Actual results

Resizing the windows keeps text hidden and you have to mouse-wheel over the text to get it to refresh.

### Expected results

After a window resize, both starting gcode, and ending gcode text elements should refresh to show newly exposed lines.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This has happened as far back as 4.7 if I remember correctly - and it makes it difficult when telling clients to add a G29 to their gcode where to put it, as they expand the window, and then don't understand why they can't see anything more.  It adds to the confusion users have when adding a BLTouch to their machine.

fvrmr, 2021-05-06:
Hi @thantik thank you for your report.
I will discuss it with the team. Keep you posted!

fvrmr, 2021-05-07:
I have discussed this with the team.
We do understand that it is a bit annoying. But we decided to defer this issue, because if you click on the text area box it will show the whole g code.
Hope you understand!

closed 2023-06-09 by github-actions[bot]