Ultimaker/Cura #9770: Printer Freeses if File Name of GCODE Contains Umlaut (non 7-Bit ASCII)

FPrefect20, 2021-05-05:
### Application Version

4.9.0

### Platform

Win 10

### Printer

Anycubic Mega S

### Reproduction steps

Not sure if this is a "bug" but since it is only a problem with CURA v.4.9.0 i report it.
Older versions of CURA (including "master") did convert filenames to the correct character set.

CURA 4.9 is not converting any longer

### Actual results

- Printer freezes (Anycubic)
- Printer msg "incomplete GCODE" (Prusa)

### Expected results

Convert filenames as in older versions of CURA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional info needed

nallath, 2021-05-05:
This was changed in https://github.com/Ultimaker/Cura/commit/fa4d1ba1fbe4685dbab21825c7fa0bcbdc889b36

Stripping it caused other issues (such as words completely changing meaning).

FPrefect20, 2021-05-06:
@fieldOfView : THANKS! 
since this may be more a "printer problem" rather than a CURA bug i will close the issue.

closed 2021-05-06 by FPrefect20