Ultimaker/Cura #9772: cannot login/connect Cura to Ultimaker Account

physi, 2021-05-05:
### Application Version

4.9.0 and 4.8.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Open Ultimaker cura,

Click on `sign in`

Website Login opens.
Enter credentials.

get redirected and this error
[image]

### Actual results

Get an error SSL (see Screenshot) ,
this is from the logs:
`2021-05-05 13:51:54,039 - INFO - [MainThread] cura.CuraApplication.writeToLog [596]: User accepted the User-Agreement.
2021-05-05 13:51:58,636 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.startAuthorizationFlow [161]: Starting new OAuth2 flow...
2021-05-05 13:51:58,638 - DEBUG - [MainThread] cura.OAuth2.LocalAuthorizationServer.start [63]: Starting local web server to handle authorization callback on port 32118
2021-05-05 13:52:04,787 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-05-05 13:52:13,772 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.startAuthorizationFlow [161]: Starting new OAuth2 flow...
2021-05-05 13:55:20,755 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.startAuthorizationFlow [161]: Starting new OAuth2 flow...`

### Expected results

Login should work

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Tested with Virusscanner disabled and with and without VPN, to exclude any local router issues.

physi, 2021-05-05:
Well I figured out, that this is a local problem. 
To solve that, I had to open `chrome://net-internals/#hsts`
and delete localhost from hsts

[image]

closed 2021-05-05 by physi

Ghostkeeper, 2021-05-17:
To communicate the log-in information to Cura, Cura sets up a local webserver temporarily listening to a specific port on connections from localhost. The website visits the localhost page with the required log-in information in its request data.