Ultimaker/Cura #9776: Print starts when the extruder is cold

kiernozekm, 2021-05-05:
### Application Version

4.9.0

### Platform

Linux Manjaro

### Printer

Monoprice Voxel

### Reproduction steps

The same STL files, on 4.8 prints well like on eralier versions, on 4.9  print starts when the bed is completely cold, same thing with an extruder. Heating doesen't start.

Startup code:
M118 X15.00 Y15.00 Z0.51 T0
M140 S{material_bed_temperature} T0
M104 S{material_print_temperature} T0
M104 S0 T1
M107
G90
G28
M132 X Y Z A B
G1 Z50.00 F400
G161 X Y F3300
M7 T0
M6 T0
M651
M907 X100 Y100 Z40 A100 B20
M108 T0

### Actual results

Bed and extruder has indoor temp.

### Expected results

Much more temp

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Files from both versions

[project_files.zip](https://github.com/Ultimaker/Cura/files/6429837/project_files.zip)

GregValiant, 2021-05-05:
M140 S{material_bed_temperature} T0
M104 S{material_print_temperature} T0
Those two commands set the temperature of the bed and hot end.  They do not wait for the temperatures to be reached before running the movement and extrusion codes that follow.  This results in the print starting before the machine heats up.
Below you see the two added lines of M190 and M109.  They are the "wait for temperature" commands.  Add them to the start-up gcode (Manage Printers / Machine Settings) immediately after the M140 and M104 commands as shown.
M140 S{material_bed_temperature} T0
M104 S{material_print_temperature} T0
M190 S{material_bed_temperature} T0
M109 S{material_print_temperature} T0

Ghostkeeper, 2021-05-17:
Indeed, M104 and M140 don't wait for the nozzle and bed to heat up, so what you're seeing would be expected behaviour of the printer then. This is just a misunderstanding of these g-codes, so I'll close this as a user error.

closed 2021-05-17 by Ghostkeeper