Ultimaker/Cura #9779: Missing support brim when using tree structure support

DanielFennhagen, 2021-05-06:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Creality3D Ender-5

### Reproduction steps

1. Insert 3d model
2. Enable tree support
3. Slice

My current settings are:

- Generate Support: True
- Support structure: Tree
- Tree support branch distance: 1.0mm
- Tree support branch diameter: 1.0mm
- Tree support Collision resolution: 0.1mm
- Support placement: Everywhere
- Support overhang angle: 45.0
- Support wall line count: 2
- Enable support bring: True
- Support brim width: 8.0mm

I'll also note that changing brim width had no effect.

### Actual results

The support did not have a brim all att locations.

See following image of layer 1 and 2:
[image]

### Expected results

Expected the support to have a brim att all starting points.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE5_keyboard.zip](https://github.com/Ultimaker/Cura/files/6432357/CE5_keyboard.zip)

fvrmr, 2021-05-06:
Hi @DanielFennhagen thank you for your report.
Try changing the `support x/y distance` to 0.4 mm. Let me know if that works for you!

GregValiant, 2021-05-06:
There are a lot of errors in the model.  Reversed surfaces and so forth.  I noticed this feature is only held to the main part by this narrow isthmus and that only exists on two layers.   The rest is just hanging in space.

[image]

Here it is again.  The red surfaces are flipped normal.

[image]

The model in the attached 3mf file has been repaired and there is a profile that works.  That hanging part that's barely connected still needs attention.
[GV_keyboard_fixed.zip](https://github.com/Ultimaker/Cura/files/6436782/GV_keyboard_fixed.zip)

DanielFennhagen, 2021-05-10:
> 
> 
> Try changing the `support x/y distance` to 0.4 mm. Let me know if that works for you!

This solution worked for me, thanks!

@GregValiant thanks for the repaired model and profile!

closed 2021-05-10 by DanielFennhagen