Ultimaker/Cura #9784: Privileges error printing from Fusion 360

FSTGuy, 2021-05-06:
### Application Version

4.9.0

### Platform

MacOS Catalina 10.15.7

### Printer

Ultimaker S3

### Reproduction steps

Error steps
1. File -> 3D Print
2. Select body
3. Press OK
4. Error
[image]

No error steps
Install 4.8 and overwrite 4.9
1. File -> 3D Print
2. Select body
3. Press OK
4. Cura 4.8 launches with no issue

Cura log file while using 4.9
[cura.log](https://github.com/Ultimaker/Cura/files/6436872/cura.log)

### Actual results

Error message
Fusion 360 does not have sufficient priveleges to launch the Print Utility. The Print Utility must be manually launched once in order to allow Fusion 360 to launch it. Please launch Print Utility from the Applications menu.

I launched the application and still encountered the issue

### Expected results

Expect to send print from Fusion 360, Cura will launch and image will appear in Cura. This is the first time I have had this issue. I have done a number of Cura upgrades in the past.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Please see attached 
1. error message
2. cura.log file.

fvrmr, 2021-05-07:
Hi @FSTGuy thank you for your report. 
I tried this myself and it worked for me. Did you update the location in Fusion when you updated to 4.9?

FSTGuy, 2021-05-07:
Hi Fenne,

Thanks for looking into my issue. I appreciate it. I installed 4.9 from Github again this morning and I encountered the same issue. Please see the errors below.

I also tried to launch Cura separately which worked with no issue. I closed Cura and tried to print from Fusion 360 and got the same error.

I have also included the Cura log.

I would be interested in your thoughts.

Thanks,

Jeremy

fieldOfView, 2021-05-07:
Unfortunately Github scrubs all attachments when you reply on an issue thread, so we did not receive your log, nor get to see your screenshots. Could you attach them via https://github.com/Ultimaker/Cura/issues/9784 instead?

Ghostkeeper, 2021-05-17:
The log file shows nothing particularly special. There is one thing of note, which may be causing that pop-up about Cura quitting unexpectedly:

```
2021-05-02 13:57:49,168 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2021-05-02 13:57:49,173 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'GlobalStack' object to a C++ 'QObject*' instance
```

This happens to me as well from time to time upon closing Cura. It seems to be something depending on the order of execution of threads. Something uses the GlobalStack while Cura is closing down and the Python object is already deleted. However it seems to be quite harmless (I do consider it more harmful if that causes that pop-up to appear on MacOS though).

However that seems to be unrelated. This is Fusion saying it is not allowed to start external applications like Cura. If it's not allowed to start Cura, there is nothing we can do at Cura to fix that.

Fusion 360 has a workaround for it which basically says what you already know: Launching Cura first works. https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/Error-Fusion-360-does-not-have-sufficient-privileges-to-launch-the-Print-Utility-trying-to-launch-Cura-from-Fusion-360-on-Mac.html

These guys also found another workaround which may be an actual fix: Select a "manual" application and then browse to Cura's path: https://forums.autodesk.com/t5/fusion-360-design-validate/3d-printing-quot-fusion-360-does-not-have-sufficient/td-p/6819297

Since this is a problem with Fusion 360, I'll close this issue.

closed 2021-05-17 by Ghostkeeper