Ultimaker/Cura #9785: (4.9.0) Skirt/Brim not using 1st layer dimensions for detecting build volume limits

jimb15122, 2021-05-06:
### Application Version

4.8.0 and 4.9.0, probably prior versions also.

### Platform

windows 10

### Printer

creality ender 3

### Reproduction steps

1. Load model with a smaller first layer than upper layers. see jpg.

2. upper layers are very close to x/y build volume limits, but not over.

3. first layer is much smaller, even with supports.  e.g. supports are not causing the problem

### Actual results

Model wont slice with brim on, but plenty of room for a brim.
[image]

### Expected results

Model should slice without turning off brim.
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[jab left gutter.zip](https://github.com/Ultimaker/Cura/files/6436966/jab.left.gutter.zip)

jimb15122, 2021-05-07:
Thanks, but this is the wrong answer.  THE MODEL AND BRIM FIT IT THE BUILD PLATE!, Perhaps a fix would be to provide an option to disable all model boundary checking, the model will slice correctly.

GregValiant, 2021-05-10:
I see no reason to yell.  It may not be the answer you wanted to hear but it's the correct answer.  Cura uses the "bounding box" to determine the model size as opposed to "footprint".  There are things I do with my own Ender that involve increasing the build plate size and adjusting the Home Offsets to compensate for this.  The alternative is to use Prusaslicer which does appear to use "footprint".

jimb15122, 2021-05-10:
Sorry for yelling.  Being an old software engineer who started in the early 1970's, I sometimes get impatent. Let me try to explain better.  I totally agree that the bounding box for the MODEL ITSELF should be based on max dims of all layers. That is obvious.  But the same is true for the bounding box for all other components that are added to the model to enable actual printing.  It is incorrect to assume that the brim will go around the widest part of the model.  A separate (2nd) bounding box should be computed using the ACTUAL dimensions of the brim, supports, etc. The decision to slice should then be based on all boxes fitting in the build space.  It is not rare for the first layer to NOT be the widest part of the model.  Consider this use case:  printing a maximal sized sphere to replace my aging lawn ball.  Even with supports set to 45 degrees, the first layer will be much smaller than the sphere diameter.  There is plenty of room for a brim, but if I enable it, cura will refuse to slice it.  Resorting to hacks to get it to slice is not ideal, especially when this can be fixed by using proper logic.

Ghostkeeper, 2021-05-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/3365.

The issue is indeed that knowing the first layer requires slicing. Sometimes it requires slicing of the entire model, e.g. if the overhanging part needs support which may then need a brim around it again which could go outside of the build volume, so just the first layer is not sufficient.

We've rejected this as a won't-fix because of the complexity involved in making this case work.

closed 2021-05-17 by Ghostkeeper

jimb15122, 2021-05-22:
Thanks for confirming this is a bug.  Given the numerous ways this bug manifests itself, as evidenced by the multiple reports related to the same fundamental problem, I hope this is addressed someday.  I'm not sure why all bounds checking (except for the model proper) can't be postponed until after slicing.  I hope it is not a matter of prioritizing speed over correctness (common in early days of computing when human costs were far less than hardware costs), but certainly not true today.

Ghostkeeper, 2021-06-04:
This is only an issue when slicing large objects. Slicing large objects takes a lot of time. Usually a few minutes. If you're unsure if the object will fit on the build plate, or fiddling with multiple objects, it's nice to get immediate feedback instead of having to wait for minutes to get this feedback. This is especially relevant for tree support, which can take up to half an hour before you'd be able to see that the model is 0.5mm too far to the edge.