Ultimaker/Cura #9790: Setting Visibility dialog is slooow to scroll when "all" is checked

GregValiant, 2021-05-07:
### Application Version

4.9.0 only (previous and Arachne are OK)

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

1. Open the "Settings Visibility" dialog.
2. Check "All".
3. Scroll - the dialog moves 2 or 3 lines and becomes unresponsive for 40 to 60 seconds.  Scrolling moves another 2 or 3 lines and it becomes unresponsive for another 40 to 60 seconds.  This happens with the scroll button, click in the "gross movement" area, or clicking on the Up/Down buttons.
4. It does work when no settings are selected and gets progressively worse as more settings are checked.  Setting the visibility to "Expert" is almost acceptable and a lot better than "All".
5. De-selecting "All" and the dialog reverts to "Custom" and works dependent on the number of settings checked.
5. With "All" selected - attempting to close the dialog results in "not responding" and it takes another minute to close.
6. The rest of Cura operates as it should.

### Actual results

With "All" checked the dialog is not useable.

### Expected results

The dialog should work as it does in other versions of Cura.  Here is a sceenshot of the problem.
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

The problem occurs with any model and with no model.

[cura.log](https://github.com/Ultimaker/Cura/files/6442764/cura.log)

fvrmr, 2021-05-10:
Hi @GregValiant thank you for your bug report.
I tried to reproduce this on Windows and @Ghostkeeper tried it on Linux and both we couldn't reproduce this. 
In your log file we couldn't find anything strange. 

So for now we will keep this issue open and wait for more information (from other users) to debug this issue.

konskarm, 2021-05-11:
@GregValiant I cannot reproduce the slow behavior but I fixed the warning in https://github.com/Ultimaker/Cura/commit/be74d4dd3f7606a63594ec72d3b82e15f9bd4264. I am not sure whether this will fix your issue though.

GregValiant, 2021-05-11:
Thanks for looking again.  I'll close this.

closed 2021-05-11 by GregValiant