Ultimaker/Cura #9792: Brim optimization

Juule08, 2021-05-08:
### Is your feature request related to a problem?

I´m trying to print an object that has engraved letters on its bottom. Due to the shape of this object (It´s an "L"), I have to print it with brim so that it doesn´t warps. The problem I founded is that CURA is generating brim inside of the engraved letters, and due to the letters thickness It´s impossible to remove the brim. 
I send 2 images and a .3mf proyect file
[image]
[image]
[PPLA-028-60_Porta Bases_v2.0.zip](https://github.com/Ultimaker/Cura/files/6445102/PPLA-028-60_Porta.Bases_v2.0.zip)

### Describe the solution you'd like

I´d like the slicer not to generate brim inside of the main perimeter of the object.

### Describe alternatives you've considered

An option to disable this.  Or an option to enable inner brims. Or a mínimum space needed for the slicer to generate a brim.

### Affected users and/or printers

Every user that tries to use brims with objects that have engraved shapes on the bed side (like the letter "B" or "O" shape)

Juule08, 2021-05-08:
I already check if brim only outside was deactivated. But it was activated. The problem is with the letters "O", "D", "B", "P", "R" and "A", as this letters have and empty space on its center which is not engraved on the object. So, the slicer detecs this and generate ouside brim in these central areas of these letters specifically.

fvrmr, 2021-05-10:
Hi @Juule08 thank you for your feature request. 
This is a duplicate of #4222 that's why I will close this issue.

closed 2021-05-10 by fvrmr