Ultimaker/Cura #9793: Arange all models bug

vsamotskiy, 2021-05-08:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Add some amount of models which is close to free space (I have 9 models)
2. Press "Arange all models" (Ctrl + R)

### Actual results

1. "Can't find Location" warning message is displayed
[image]
2. Not all models based on build plate
[image]
Only 8 models based on the build plate but there is empty space for 9th model exists.

### Expected results

All models based on build plate
Here is the picture that all models can be based on build plate:
[image]
9 models based manually at build plate to display availability

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6445463/cura.log)

[Project file.zip](https://github.com/Ultimaker/Cura/files/6445460/Project.file.zip)

fvrmr, 2021-05-10:
Hi @vsamotskiy thank you for your report.
We are aware of this issue and it is happening because of the library we use for the algorithm for arranging models. This is based on a convex hull. 
So we can't really change this, that's why I will defer this issue. 
Hope you understand.

Ghostkeeper, 2021-05-17:
The library decides on the optimal order in which to place objects, and then also on the optimal position to place that model. And continues with that for the next model. The problem here is that this goes into a local optimum where the placement of the 8th model is such that the convex hull of all models is minimised. But this local optimum doesn't allow for enough space to place the 9th model.

3DEnthusiast, 2022-05-02:
> Hi @vsamotskiy thank you for your report. We are aware of this issue and it is happening because of the library we use for the algorithm for arranging models. This is based on a convex hull. So we can't really change this, that's why I will defer this issue. Hope you understand.

So it would seem to me that if this is an issue with the library then perhaps you need to rebuild a new library. Honestly, every other slicer can handle this function. I fail to understand why this is still an issue even into version 5 beta.

The slicing engine has been changed and reworked for V5. It's time to point the focus onto an issue that all the users have mentioned time and time again.

batesman1, 2022-06-22:
[image]
[image]
[image]

3DEnthusiast, 2022-06-22:
> [image] [image] [image]

Yeah, good luck. Ultimaker doesn't apparently care to fix this.

tastyratz, 2022-06-23:
This isn't even a problem limited to convex hull for build plates with many objects.

Some of this could be models built or saved off their centerline. If that's the case, why wouldn't Cura calculate object center before determining positioning instead of model file center?

More often than not when I drag a single STL onto the window it appears like the example above, way off the side of the plate. I don't think that this is the entire problem but I can't imagine why the first object on an empty plate shouldn't always appear center plate to start?

Maybe whatever is causing that is also messing with convex hull?

HOLLY-DION, 2022-06-25:
When I want to play a whole tray of different miniatures, 5.0 does not fill my 220cm * 220cm panel well. Can not reasonably use the space, has been arranged in the center, but around is some unusable area, if in a reasonable placement, is still able to put a few more models.

glazedfaith, 2023-01-07:
How is this still a problem 18 months on?

s9967839, 2023-01-09:
Yeah, Im still getting this issue with square items on the current latest version of cura

s9967839, 2023-01-09:
Just a quick update, when you turn off "Support material -> Skirt" and auto arrange, this fixes the problem. Then you can re-enable skirt.

aswyatt, 2023-01-22:
I'm having the same problem: Cura cannot place multiple models even when they can clearly fit. I would understand if I had lots of items and it couldn't figure out the last piece like in the original post, but even when there are only two items it doesn't work. The fact that turning off the skirt allows it to fit clearly indicates a more basic problem than described. What's even more galling is that it used to work fine. It's great that Cura is being updated, but one should ensure working functionality does not get broken.

Side note: when I removed skirt and auto-placed (worked), Cura moved the two models onto the build plate, but still thought the bounding box of one of the models was outside the build plate (model was still hatched and the move cursors in the original location). Repeating the process then worked.
[image]

fieldOfView, 2023-01-22:
It looks like in the specific case in the image by @aswyatt , Cura does not properly get the bounding box of the model. This is evidenced by the move widget axes not being in the center of the model. It would be interesting to see the original 3mf files.

Using stl files instead of 3mf files would solve this problem, because the origin and bounding box are always reset when loading an STL file.

aswyatt, 2023-01-22:
> It looks like in the specific case in the image by @aswyatt , Cura does not properly get the bounding box of the model. This is evidenced by the move widget axes not being in the center of the [...]
[image]

Unfortunately not. Indeed there is a problem getting the bounding box correctly with 3ml files - it doesn't automatically centre the model on the build plate when loading, but does centre when selecting arrange models (annoying quirk which should be fixable). With STL files, it does autocentre when loading. However, if I moved one of the models off the build plate, multiple, auto-arrange, then it still thinks the bounding box is outside (this is from an STL file).

closed 2024-12-09 by github-actions[bot]