Ultimaker/Cura #9794: Suggestions on setting values of variables

xsuper9988, 2021-05-08:
### Is your feature request related to a problem?

When i use the Variables in start G-code，and My "Inital Fan Speed = 60%"

M106 P{cool_fan_speed_0}

Then i get M106 P60
This is OK in PLA

When i use ABS，and i "Disable Cooling

Then i still get M106 P60
This is not OK in ABS

### Describe the solution you'd like

When i "Disable Cooling

I think "cool_fan_speed_0" should be 0
M106 P0

### Describe alternatives you've considered

NO

### Affected users and/or printers

ABS user
or High temperature materials

GregValiant, 2021-05-08:
What printer and firmware?
[MarlinFW.org](https://marlinfw.org/docs/gcode/M106.html) says that M106 P indicates which fan.  M106 S indicates the fan speed as PMW (0 to 255).  The layer cooling fan at 60% should be M106 S153.  If the printer has the cooling fan(s) on a single circuit there should be no reason to use M106 P.

xsuper9988, 2021-05-11:
> What printer and firmware?
> [MarlinFW.org](https://marlinfw.org/docs/gcode/M106.html) says that M106 P indicates which fan. M106 S indicates the fan speed as PWM (0 to 255). The layer cooling fan at 60% should be M106 S153. If the printer has the cooling fan(s) on a single circuit there should be no reason to use M106 P.

Sorry
I always used "S" before

Because i use this in "Start G-Code"
M106 S{cool_fan_speed_0}

And CURA4.9 slice result is
M106 S60

Then i think "P" refers to percentage
And i have not tested the P effect

xsuper9988, 2021-05-11:
\

> Hi @xsuper9988 thank you for your request.
> This is expected behaviour, the settings are not compatible with the start gcode, it means that the settings can't overwrite the start gcode. So if you have `M106 P{cool_fan_speed_0}` in your start gcode it will always be used even if you disable cooling.

Maybe #9795 solved
There is no problem
Because I don't need to increase "Start G-code" by myself

xsuper9988, 2021-05-11:
Sorry everyone
i test delete
M106 S{cool_fan_speed_0}

and slice result is 

;FLAVOR:Marlin
;TIME:1981
;Filament used: 1.54576m
;Layer height: 0.2
;MINX:74.348
;MINY:85.16
;MINZ:0.2
;MAXX:104.028
;MAXY:114.84
;MAXZ:20
;ARCWELDERPROCESSED
; Postprocessed by [ArcWelder](https://github.com/FormerLurker/ArcWelderLib)
; Copyright(C) 2020 - Brad Hochgesang
; resolution=0.05mm
; path_tolerance=1%
; max_radius=1000000.00mm
; firmware_compensation=True
; mm_per_arc_segment=1.00mm
; min_arc_segments=12
; default_xyz_precision=3
; default_e_precision=5

;Generated with Cura_SteamEngine 4.9.0
M104 S240
M105
M109 S240
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M104 S240 ;Start heating extruder
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S240 ;Wait for extruder to reach temp before proceeding
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5.0 Y20 Z0.3 F5000.0 ; Move to start position
G1 X5.0 Y150.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X5.2 Y150.0 Z0.3 F5000.0 ; Move to side a little
G1 X5.2 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F2100 E-5
;LAYER_COUNT:100
;LAYER:0
M106 S153

I think my "Feature Request Title" should chang "M106  add option at start or not"
Because when my hot end is preheat finish
But the fan(M106) behind hot end
It will cause hot end temperature instability
Even if it's a small problem

fvrmr, 2021-05-20:
We discussed this issue with the team. And decided to defer this issue, because it is not relevant to Ultimaker printers. 
To get all the settings to work is out of scope.
Like you can read above `cool_fan_speed` is not compatible in the start gcode. 
I hope you understand!