Ultimaker/Cura #9795: About M109 "Ahead of G29" or "Behind G29" Option Setting

xsuper9988, 2021-05-08:
### Is your feature request related to a problem?

Because "25 points" and "Hot End" take a lot of time

### Describe the solution you'd like

when i use 25 points ABL
and i didnt use Hot Bed
I hope M109 "Behind G29"
when i finish ABL
Hot end is almost done

when i use ABL
and i use Hot Bed
and The hot bed may warp
then i hope wait for Hot bed finish
os i hope M109 "Ahead of G29"

### Describe alternatives you've considered

NO

### Affected users and/or printers

when anyone dont use hot bed

GregValiant, 2021-05-09:
The G29 command is usually added from the StartUp G-Code in your machine settings.  It would be helpful if you would post your startup G-code.

fvrmr, 2021-05-10:
Hi @xsuper9988 thank you for your request.

Like @GregValiant mentioned a sharing your start gcode would be helpful. 

In the settings guide you can find more information about heating befor the start g-code. https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_start_gcode.md#heating-before-the-start-g-code

xsuper9988, 2021-05-11:
> @xsuper9988 you can use variables in your "Start GCODE" and do the heating yourself. Cura doesn't add heating when you already did:
> [image]

Thank you
But it doesn't seem to be the case
My StartG-code On TOP
And Cura4.9 Slice result is

;FLAVOR:Marlin
;TIME:1982
;Filament used: 1.5458m
;Layer height: 0.2
;MINX:95.16
;MINY:85.16
;MINZ:0.2
;MAXX:124.84
;MAXY:114.84
;MAXZ:20
;ARCWELDERPROCESSED
; Postprocessed by [ArcWelder](https://github.com/FormerLurker/ArcWelderLib)
; Copyright(C) 2020 - Brad Hochgesang
; resolution=0.05mm
; path_tolerance=1%
; max_radius=1000000.00mm
; firmware_compensation=True
; mm_per_arc_segment=1.00mm
; min_arc_segments=12
; default_xyz_precision=3
; default_e_precision=5

;Generated with Cura_SteamEngine 4.9.0
M104 S240
M105
M109 S240
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M104 S240 ;Start heating extruder
M106 S60
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S240 ;Wait for extruder to reach temp before proceeding
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5.0 Y20 Z0.3 F5000.0 ; Move to start position
G1 X5.0 Y150.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X5.2 Y150.0 Z0.3 F5000.0 ; Move to side a little
G1 X5.2 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

M104 S240
M105
M109 S240
These are duplicated

And i have another Feature Request #9794 
Related to the request here (#9795)

d-schmidt, 2021-05-11:
Try also heating the bed in you custom start gcode. This feature is not documented,  I think, so you have to try.
My Cura honors my custom heating:
[image]

GregValiant, 2021-05-12:
I'm confused.
M104 is set Hot End Temperature and M109 is wait for Hot End to reach Temperature.
M140 is set Bed Temp and M190 is Wait for Bed to reach Temperature.
So:
If the bed is off you want ABL with the Hot End off also ("I hope M109 "Behind G29"")
M190 S0  ;bed off
G29  ;ABL
M109 S240  ;wait for hot end to hit 240

if the bed is on you want ABL with both the bed and hot end completely heated ("i hope M109 "Ahead of G29"")
M190 S70 ;wait for bed before continuing
M109 S240 ;wait for hot end before continuing
G29  ;ABL

Is that the kind of thing you are looking for?

d-schmidt, 2021-05-12:
I heat my bed before the ABL and then heat the extruder while ABL.  Because the extruder temperature doesn't matter for ABL, it shortens print time and leaving the extruder at high temps without printing for long amounts of time isn't good

If you don't want the bed to heat you probably need in the start code to trick cura into not adding heating again. I think adding M140 M190 even if you don't want to heat your bed should work. If they are set to 0 they should be skipped.

xsuper9988, 2021-05-13:
> I'm confused.
> M104 is set Hot End Temperature and M109 is wait for Hot End to reach Temperature.
> M140 is set Bed Temp and M190 is Wait for Bed to reach Temperature.
> So:
> If the bed is off you want [...]

YES
When not using hot bed
I can start ABL and HOT END together
This can saves time

xsuper9988, 2021-05-13:
> I heat my bed before the ABL and then heat the extruder while ABL. Because the extruder temperature doesn't matter for ABL, it shortens print time and leaving the extruder at high temps without printing for long amounts of time isn't good.
> 
> If you don't want the bed to heat you probably still need M190 in the start code to trick cura into not adding heating again. I think adding M140 M190 even if you don't want to heat your bed should work. If they are set to 0 they should be skipped.
> add:
> M140 S0
> M190 S0

OK
That is good idea
Thank you

xsuper9988, 2021-05-13:
> I heat my bed before the ABL and then heat the extruder while ABL. Because the extruder temperature doesn't matter for ABL, it shortens print time and leaving the extruder at high temps without [...]

I tested start g-code

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

M104 S{material_print_temperature_layer_0 } ;Start heating extruder
M140 S0
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S{material_print_temperature_layer_0 } ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

and get slice result 

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

;FLAVOR:Marlin
;TIME:2305
;Filament used: 1.67648m
;Layer height: 0.2
;MINX:89
;MINY:79
;MINZ:0.2
;MAXX:131
;MAXY:121
;MAXZ:20
;Generated with Cura_SteamEngine 4.9.1
M104 S240
M105
M109 S240
M82 ;absolute extrusion mode
M104 S240.0 ;Start heating extruder
M140 S0
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S240.0 ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

M104 S240
M105
M109 S240
M82 ;absolute extrusion mode

These codes are still repeated

xsuper9988, 2021-05-13:
> I tested start g-code
>
> =-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
>
> M104 S{material_print_temperature_layer_0 } ;Start heating extruder
> M140 S0
> M106 S153
> G92 E0 ; Reset Extruder
> G28 [...]

@fvrmr These codes are still repeated

xsuper9988, 2021-05-13:
> I tested start g-code
>
> =-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
>
> M104 S{material_print_temperature_layer_0 } ;Start heating extruder
> M140 S0
> M106 S153
> G92 E0 ; Reset Extruder
> G28 [...]

@fvrmr Is this a bug?

d-schmidt, 2021-05-16:
cura adds M105 between M104 and M109, maybe try adding that manually in your start as well.

xsuper9988, 2021-05-17:
> cura adds M105 between M104 and M109 for your printer, maybe try adding that manually in your start as well.

M104 S240
M105
M109 S240
M82 ;absolute extrusion mode

These codes are still repeated

Satrt G-Code
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

M104 S{material_print_temperature_layer_0 } ;Start heating extruder
M140 S0
M105
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S{material_print_temperature_layer_0 } ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

Cura 4.9.1 Slice Result
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

;FLAVOR:Marlin
;TIME:2305
;Filament used: 1.67648m
;Layer height: 0.2
;MINX:89
;MINY:79
;MINZ:0.2
;MAXX:131
;MAXY:121
;MAXZ:20
;Generated with Cura_SteamEngine 4.9.1
M104 S240
M105
M109 S240
M82 ;absolute extrusion mode
M104 S240.0 ;Start heating extruder
M140 S0
M105
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S240.0 ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

d-schmidt, 2021-05-17:
Okay I'm out if ideas why it doesn't work for you. Don't really want to read the code to understand the problem.

You can always just open the gcode file and delete them using a good text editor like notepad++, sublime, atom, any ide

Ghostkeeper, 2021-05-17:
I'm unable to reproduce this bug. I copied your start g-code into a new Ender 3 printer, but removed the extraneous spaces in your setting keys (don't know why you added those?)

```
M104 S{material_print_temperature_layer_0} ;Start heating extruder
M140 S0
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

The slice result was then:

```
;FLAVOR:Marlin
;TIME:561
;Filament used: 0.233232m
;Layer height: 0.2
;MINX:101.5
;MINY:101.5
;MINZ:0.2
;MAXX:133.5
;MAXY:133.5
;MAXZ:10
;Generated with Cura_SteamEngine master
M140 S50
M105
M190 S50
M82 ;absolute extrusion mode
M104 S200 ;Start heating extruder
M140 S0
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S200 ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:50
;LAYER:0
```

Could you try removing the extra spaces from your setting keys? They do not belong there and might be causing this issue.

d-schmidt, 2021-05-17:
@Ghostkeeper you still heat the bed twice. I guess you need to use variables for all 4 temperature settings:
[image]
works for me

xsuper9988, 2021-05-18:
> @Ghostkeeper you still heat the bed twice. I guess you need to use variables for all 4 temperature settings:
> [image]
> works for me. Set the bed to 0 in print quality settings.
> 
> ```
> ; custom heating
> M140 S{material_bed_temperature_layer_0} ; Start heating bed
> M105
> M106 S127 ; extruder fan 50%
> M104 S{material_print_temperature_layer_0} ; Start heating extruder
> G28 ; Home all axes
> G29 ; ABL
> M190 S{material_bed_temperature_layer_0} ; Wait for bed temp
> M109 S{material_print_temperature_layer_0} ; Wait for extruder temp
> 
> ; Ender 3 Custom Start G-code - Start
> G92 E0 ; Reset Extruder
> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
> G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
> G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
> G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
> G92 E0 ; Reset Extruder
> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
> ; Ender 3 Custom Start G-code - End
> ```

I delete M140、G92、M82、G190
And then it's OK
I tested that I only deleted G92、M82
And it's also OK

Abnormal Satrt G-Code
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

M104 S{material_print_temperature_layer_0 } ;Start heating extruder
M140 S0
M105
M106 S153
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Auto bed level
M109 S{material_print_temperature_layer_0 } ;Wait for extruder to reach temp before proceeding
M190 S0
M82 ;absolute extrusion mode

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

Normal Satrt G-Code
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

M104 S{material_print_temperature_layer_0} ;Start heating extruder
M106 S153
G28 ; Home all axes
G29 ; Auto bed level
M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach temp before proceeding

; Ender 3 Custom Start G-code
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

I don't know why this is a problem(G92、M82)
@d-schmidt 

@fvrmr 
And then please evaluate this Function
(About M109 "Ahead of G29" or "Behind G29" Option Setting)

Thank you very much everyone

xsuper9988, 2021-05-18:
> I'm unable to reproduce this bug. I copied your start g-code into a new Ender 3 printer, but removed the extraneous spaces in your setting keys (don't know why you added those?)
> 
> ```
> M104 S{material_print_temperature_layer_0} ;Start heating extruder
> M140 S0
> M106 S153
> G92 E0 ; Reset Extruder
> G28 ; Home all axes
> G29 ; Auto bed level
> M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach temp before proceeding
> M190 S0
> M82 ;absolute extrusion mode
> 
> ; Ender 3 Custom Start G-code
> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
> G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
> G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
> G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
> G92 E0 ; Reset Extruder
> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
> ```
> 
> The slice result was then:
> 
> ```
> ;FLAVOR:Marlin
> ;TIME:561
> ;Filament used: 0.233232m
> ;Layer height: 0.2
> ;MINX:101.5
> ;MINY:101.5
> ;MINZ:0.2
> ;MAXX:133.5
> ;MAXY:133.5
> ;MAXZ:10
> ;Generated with Cura_SteamEngine master
> M140 S50
> M105
> M190 S50
> M82 ;absolute extrusion mode
> M104 S200 ;Start heating extruder
> M140 S0
> M106 S153
> G92 E0 ; Reset Extruder
> G28 ; Home all axes
> G29 ; Auto bed level
> M109 S200 ;Wait for extruder to reach temp before proceeding
> M190 S0
> M82 ;absolute extrusion mode
> 
> ; Ender 3 Custom Start G-code
> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
> G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
> G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
> G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
> G92 E0 ; Reset Extruder
> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
> G92 E0
> G92 E0
> G1 F2700 E-5
> ;LAYER_COUNT:50
> ;LAYER:0
> ```
> 
> So it put the build plate heating commands still before the start g-code. This is expected because your start g-code contains no settings relating to the build plate temperature, only to the nozzle temperature.
> 
> Could you try removing the extra spaces from your setting keys? They do not belong there and might be causing this issue.

Thank you I finally confirmed the real problem

@d-schmidt problem is extra spaces

@fvrmr
And then please evaluate this Function
(About M109 "Ahead of G29" or "Behind G29" Option Setting)

Thank you very much everyone

Ghostkeeper, 2021-05-19:
There are a lot of possible start g-codes, necessary for all of the hundreds of printers on the market. We're not going to provide options to automatically generate all sorts of combinations of them. That's why the start g-code is a free-form text field. So we'll reject this feature request.

Cura currently doesn't even generate the `G29` itself; that's all in your start g-code.

closed 2021-05-19 by Ghostkeeper