Ultimaker/Cura #9797: Combing mode Problem

gottabigproblem, 2021-05-09:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3 pro

### Reproduction steps

Changed combing mode to all
Then, put the max distance with no retract at 2
[image]
00mm
[image]
[image]

### Actual results

There was still stringing

### Expected results

There should have been no stringing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]

GregValiant, 2021-05-09:
I think you need to look at other settings.  When you have a part that splits into separate areas (islands) as it gets taller (like a table upside down with the legs sticking up, or a horizontal through hole), those separate areas require the print head to cross a gap.  Combing can't do anything about that.  Retraction distance and speed, travel speed, print temp, maybe Z-hop, Z seam location, layer start, point...those kinds of settings have an effect.  Combing is good when all of a part is contiguous, but has no effect when crossing a gap.  The printer _must_ cross the gap and if it is oozing there's gonna be a string.  Minimizing the oozing is the key.

gottabigproblem, 2021-05-09:
im using tpu so i cant have retraction but i get what u mean

fvrmr, 2021-05-10:
Hi @gottabigproblem 
Stringing can be caused by a lot of settings. 
In the settings guide you can read more about stringing: 
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/stringing.md

If you want you could share your project file and I can take a closer look and see what you could change.

gottabigproblem, 2021-05-10:
how do i sahre my project file

fvrmr, 2021-05-11:
You can share it as a .zip in a comment here on Github. Or share a Google drive link. 

To save a project file go to file - save project.

gottabigproblem, 2021-05-11:
the

GregValiant, 2021-05-12:
I think you need to put it in a zip file.  This website doesn't accept 3mf files.

closed 2021-05-31 by no-response[bot]