Ultimaker/Cura #9798: Add Minimal Surface Area for Ironing

d-schmidt, 2021-05-09:
### Is your feature request related to a problem?

We can select top and everywhere. Everywhere does iron even miniscule invible notches wasting time:  
green areas  
[image]

### Describe the solution you'd like

I would like an option:  
Iron surface area > 4  mm²

### Describe alternatives you've considered

-

### Affected users and/or printers

all

### Additional information & file uploads

I used this in other flower pots to create water tight surfaces. But for this one it adds 20% print time.

d-schmidt, 2021-05-10:
Hi @fvrmr looks like the same problem basic problem with a different solution suggestion.

In the end, this is a waste of time:
[image]

fvrmr, 2021-05-11:
Hi @d-schmidt yes I know.
I have discussed it with the team. And we've decided to defer this issue, since it is not a priority for us to add. 
Hope you understand.

d-schmidt, 2023-07-01:
This is still relevant as Cura still irons the smallest areas:
[image]