Ultimaker/Cura #9800: Tree Support Forcing Shared Support Bases

CCS86, 2021-05-09:
### Application Version

4.9.0

### Platform

Win 10 x64

### Printer

UMO+

### Reproduction steps

Enable tree support with multiple models positioned near each other.

### Actual results

The support engine seems dead set on having a combined support base for different supported regions, even on different parts.

### Expected results

When the models are super close to each other, combined support bases can make sense, it's cool that the engine even has awareness of support structures for different parts. But, when having the bases connect actually uses more material/time, it clearly makes no sense.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

You can see in the right of this screenshot, how big of a tree support that is required for this feature. 

[image]

[SupportSharedBase.3mf.txt](https://github.com/Ultimaker/Cura/files/6448272/SupportSharedBase.3mf.txt)

fvrmr, 2021-05-10:
Hi @CCS86 thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/7797
There you can also find a work around for this issue: https://github.com/Ultimaker/Cura/issues/7797#issuecomment-633549223

I will close this issue since it is a duplicate.

closed 2021-05-10 by fvrmr

CCS86, 2021-05-10:
Why was the original issue closed? 

I see no actual solution there, just a workaround that is a bit clunky.

fvrmr, 2021-05-10:
I closed the issue since it is a duplicate of that issue I linked above.

This issue with tree support is expected behaviour, because the trees always want to connect to one base, even when the models are apart from each other.
The work around I suggested is the only way to disconnect the tree supports.

fvrmr, 2021-05-11:
The original is not closed either. There is just no resolution for this.

Like @Ghostkeeper commented on the other issue:
> Support in general is not specific to any model at the moment. Essentially CuraEngine generates one support object for the entire print, and you can adjust the settings of that object but not for the supports that happen to be underneath a certain model.
> 
> Changing that is problematic. Not just because it'd be a major refactor of basically the entire support generation stage (one of the 6 stages of the slicing process) but also because it makes supported dual extrusion models very weird.

So that's why we won't change this and why this is expected behaviour. 

Let me know if that makes sense to you.