Ultimaker/Cura #9806: Drucker GEEETECH A30T 3D in der aktuellen Version von Cura installieren. Herstellerseite A30T 3-1 .curaprofile. Drucker ist nicht in der Übersicht vorhanden

PeterB1304, 2021-05-10:
### Application Version

4.9.0

### Platform

Windows 10 Home

### Printer

nicht in der Auswahl

### Reproduction steps

Kann den Drucker nicht einbinden.

### Actual results

Drucker GEEETECH A30T 3D in der aktuellen Version von Cura installieren. Herstellerseite A30T 3-1 .curaprofile. Drucker ist nicht in der Übersicht vorhanden

### Expected results

Wie bekomme ich den Drucker installiert.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Für Hilfe wäre ich sehr dankbar.

GregValiant, 2021-05-11:
See if there is something in this thread that helps (or at least explains).
[Ultimaker Cura Forum A30T](https://community.ultimaker.com/topic/36836-manage-printers-and-renaming/?tab=comments#comment-288219)

PeterB1304, 2021-05-11:
> Hi @PeterB1304 <https://github.com/PeterB1304>  thank you for your report.
> My German is unfortunately not so good to completely understand your issue. Please fill in the form in English.
> What I understood is that you can't install the Geeetech A30T.
> The Geeetech A30T is not in Cura as a printer, you can add it as a custom printer.
>
> Could you share your log file and curaprofile?
> You can find your log file here: C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
>
>
> This email has been scanned by BullGuard antivirus protection.
> For more info visit www.bullguard.com <http://www.bullguard.com/tracking.aspx?affiliate=bullguard&buyaffiliate=smtp&url=/>

Hello, 

i've already tried to install the A30T myself using the custom function, but the values for the A30T in the Profiles are wrong. 

 

With this mail I am sending you my CURA LOG.FILE. 

 

Perhaps you have the opportunity to send me the appropriate A30T driver for the CURA software package. 

 

With best regards Peter Buegel

 

 

: Tennisweg 30, 38667 Bad Harzburg

E-Mail:      ***@***.*** ***@***.***> 

Phone:     +49 (0) 5322 87 86 85

Mobil:       +49 (0) 151 11 80 10 37

fvrmr, 2021-05-11:
Hi @PeterB1304 
I can't see your log file, because you've replied via email.

Please attach the log file and the cura profile in a comment on Github in this thread. 

And we don't maintain the profiles and printer settings voor Geeetech printers our self.

PeterB1304, 2021-05-11:
Hello,
I have attached my LOG file and hope that you can open it.

Kind regards

PeterB1304, 2021-05-11:
Hello,
found another name in C: \ Program Files \ Ultimaker Cura 4.9.0 \ resources \ extruders folder, geeetech_A20T_1.def.json / geeetech_A20T_2.def.json / geeetech_A20T_3.def.json.

I made a COPY of this and renamed it to geeetech_A30T_1.def.json / geeetech_A30T_2.def.json / geeetech_A30T_3.def.json.
But the Ducker after restarting CURA 4.9.0 these are not available in the selection list.
Everything as Greg Valiant described.

I have no more ideas and ask for help that can also solve my problem. I need the correct settings for the Geeetech A30T printer so that it can print properly.

I'm the only one who has these problems, it's hard to imagine.

The manufacturer only sells older items for version 4.4.1. Unfortunately, these don't help me because I have the current version 4.9.0.

With friendly greetings
P. Buegel
 <https://user-images.githubusercontent.com/83987822/117845613-02a7aa80-b281-11eb-8fb8-003224e8cd9e.png>

GregValiant, 2021-05-11:
There are extruder files and there are Definition files.  I thought that post was clear that you need to make a copy of the definition file.  On my computer it is located in "C:\Program Files\Ultimaker Cura 4.9.0\resources\**definitions**" and the file is named "geeetech_A20T.def.json".  The definition and extruder files come from the printer manufacturer and are submitted to Ultimaker for inclusion in Cura.  If you require a definition file for your A30T you need to get it from Geeetech.

PeterB1304, 2021-05-12:
Hello Mr. Gerg Valiant, 

thank you for your message. 

 

I found this file "geeetech_A20T.def.json" on my system in this folder and renamed it to "geeetech_A30T.def.json" with an editor 1: 1 can be adopted for the A30T in terms of printing properties and parameters. 

 

I changed the dimensions for the print size to 320x320x420mm. 

 

Do the other parameters of the A20T still match the changed print size 320x320x420mm for the A30T. 

 

Do you have experience with printing yourself whether the parameters are correct when printing model objects. 

 

Otherwise, unnecessary materials will be printed with which you cannot start. 

 

It would be very nice if you could help me with this. 

 

The settings with the changed print size and the extruder 0/1/2 must be adjusted from the file "geeetech_A20T.def.json". which was renamed to "geeetech_A30T.def.json". 

 

If you need a platform entry in a mesh data file for the A30T, this would be advantageous for the Ducker A30T. 

 

Since I'm only just starting out with 3D printing, I don't have much experience with 3D printing yet. 

 

I just hope that I have chosen this Geeetech A30T well in terms of price / performance / pressure range and print head with 3 in 1. 

 

Or do you have another purchase recommendation for a 3D printer that has even better properties. 

 

Sincerely yours 

 

P. Buegel

 

 

 

 

Mit freundlichen Grüßen

 

Peter Buegel 

 

                

Kontakt:   Peter Buegel

Adresse:  Tennisweg 30, 38667 Bad Harzburg

* E-Mail:                    ***@***.*** ***@***.***> 

* E-Mail:                    ***@***.*** ***@***.***> 

* E-Mail:                    ***@***.*** ***@***.***> 

 

* Phone:                   +49 (0) 5322 878685 

* Fax:                        +49 (0) 5322 878677

* iPhone-Mobil:          +49 (0) 151 11801037

GregValiant, 2021-05-12:
Just post a reply here. There is no need to email.
I don't have one of those printers.  It appears they are fine.
Simple is better.  There is much to learn.  A single extruder printer has a steep learning curve by itself.  Starting with a 3 extruder machine is...tougher

PeterB1304, 2021-05-12:
Hello Mr. Greg Valiant,
Thank you for your message.

I received an email from Geeetech today with an attachment.
A file for an A10T0.2 PLA packed as RAR, the end packed is the name A10T 0.2 PLA.curaprofile

Unfortunately, I can't edit this file, you can adapt it to me for the A30T.

There was also a PDF file Configuring Cura 4-A10M.pdf for a printer other than the A30T.

Can you please take a look at the document?

Maybe other users also have these problems like me and you can also help other users with them.

With best regards
P. Buegel 
[A10T 0.2 PLA.zip](https://github.com/Ultimaker/Cura/files/6467740/A10T.0.2.PLA.zip)
[Configuring Cura 4-A10M.pdf](https://github.com/Ultimaker/Cura/files/6467743/Configuring.Cura.4-A10M.pdf)

PeterB1304, 2021-05-12:
Hello Mr. Greg Valiant,
Thank you for your message.

I received an email from Geeetech today with an attachment.
A file for an A10T0.2 PLA packed as RAR, the end packed is the name A10T 0.2 PLA.curaprofile

Unfortunately, I can't edit this file, you can adapt it to me for the A30T.

There was also a PDF file Configuring Cura 4-A10M.pdf for a printer other than the A30T.

Can you please take a look at the document?

Maybe other users also have these problems like me and you can also help other users with them.

With best regards
P. Buegel

GregValiant, 2021-05-12:
I can't do technical support for printers I know nothing about and I have no inclination to do technical support for printers I do know something about.  That's what retirement is for...to only do what I feel like doing.

Ghostkeeper, 2021-05-19:
And we don't provide technical support for modding either. This is not a bug, just a broken mod.

Your screenshots reveal that the new files still contain data and references to the A20T rather than the A30T. You will also have to edit the contents of the files, not just the file names, to change references to other A20T profiles into references to the new A30T profiles.

The PDF simply instructs you to use Custom FFF Printer, for which you don't need to edit any files. That is a much more sensible approach than making your own Cura mod, if you're not so technically inclined. We also have [this wiki page](https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura) with some explanations if you are.

This is currently not a bug. It's expected that there is no A30T profile as long as it's not contributed to us. So I'll close this issue.

closed 2021-05-19 by Ghostkeeper