Ultimaker/Cura #9810: Defect on cylinder with 3 walls

zenturacp, 2021-05-11:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

When i print stuff with 3 walls i get a pattern like this on the skin,

Its only at 3 walls, not if the item is solid, then its perfect as you can see on the bottom so its not a setting in the printer, if i print the cylinder as solid it comes out very nice, with no defects at all.

As you can see the defect is in the print its not a string, the pattern is almost consistent, and as i have highlighted there is a recurring mark on the side (The red boxes)

It happens on all circle

Its also happening if i produce a 3 walls straight (50mm/20mm /  1,2mm wide with fillets)

### Actual results

I get this pattern

[image]

### Expected results

I'm expecting this to be a nice looking consistent seam

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The defect is not shown in the preview

closed 2021-05-11 by fvrmr

zenturacp, 2021-05-11:
Okay get you - though the issue is not in 4.8 or Prusa Slicer.. Just appeared in 4.9..

Will test further to see if its 4.9 and what the difference is in the GCode from 4.8 to 4.9

fvrmr, 2021-05-11:
Did you use the same profiles in 4.8 and 4.9?

zenturacp, 2021-05-12:
I can now confirm the issue is also on Ultimaker machines

But i have tested it with the new Cura Arcane Beta, and here its sliced as it should be - also same with PrusaSlicer.

There is a supplied STL here you can test with - as you can see it runs 4 times pr. layer.

https://youtu.be/cgGcDqso0Dk i have composed a video of the issue and attached the STL file that i test with

[Test3WallsSTL.zip](https://github.com/Ultimaker/Cura/files/6466099/Test3WallsSTL.zip)

fvrmr, 2021-05-17:
Hi @zenturacp thank you for investigating it further. 
But it seems like this is a duplicate then of https://github.com/Ultimaker/Cura/issues/9356
As you can read there it is fixed on Arachne but Arachne is not implemented in 4.9 so that's why it is happening there and not in Arachne.

zenturacp, 2021-05-17:
Yes it's exactly the same issue - I did not find the other issue (was closed) probably why my search did not tell it..

Also tested in the new beta and can confirm it's fixed