Ultimaker/Cura #9813: Model preview is broken when moved between monitors.

Hajto, 2021-05-11:
### Application Version

4.9.0

### Platform

Mac OS

### Printer

Ender 3 V2 modded (not applicable though)

### Reproduction steps

Open Cura on ultra wide monitor full size window.
Add model
Slice
Send to printer via octo print
move window to second monitor

### Actual results

plate orbit pivot point is wrong. The build plate model is in the screen corner and when orbiting, it orbits around the build plate which is missplaced in the corner. Resizing the window fixes the issue.

### Expected results

build plate is in the center of the window

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://www.youtube.com/watch?v=vzMBeXS-yoI

Hajto, 2021-05-11:
[logs.log](https://github.com/Ultimaker/Cura/files/6460691/logs.log)

Ghostkeeper, 2021-05-19:
The logs don't reveal anything that could be causing this as far as I could see, sadly. They might have, but alas.

Looking at that video, I think it might occur when the other monitor has a different pixel scaling than the original monitor. I don't currently have a different monitor to test with. I can't reproduce this on Linux by just changing the pixel scale in my monitor control panel either.

I'll ask if one of our testers can try to reproduce this issue on MacOS with a second window.

fvrmr, 2021-05-20:
We have discussed this with the team and decided to defer this issue for now. But we will keep an eye out for this sort of issues.

AndreaCCIE, 2021-05-27:
I'm having the same problem on a multi-screen MacBook Pro. I can only use the integrated screen, otherwise the pivot is not centred.
The odd part is that if I uninstall Cura, remove all its folders, and install it again it works properly for a while. After some days, the issue appears again.

closed 2023-06-09 by github-actions[bot]