Ultimaker/Cura #9814: Ventilador de capa ne se activa

MikeMaker3D, 2021-05-12:
### Application Version

4.9.1

### Platform

windows 10

### Printer

Ender 3 

### Reproduction steps

El ventilador de capa no se activa aunque este la pestaña de refrigeración de capa activada.

### Actual results

Modelos impresos deformados por exceso de temperatura

### Expected results

Modelos correctamente enfriados con 0 deformaciones.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2021-05-13:
If you would put the gcode file in a zip file and use the File | Save Project command to make a 3mf file with your model and settings, and also put it into the zip file and then post it here that would be good.

DjSteker, 2021-05-19:
Puedes probar M106 P0 S128 
M106 P(FanIndex)s S(speed)s   

y si no funciona quizás tenga algún fallo la impresora como un cable , a mi me dio un error cuando actualice a la versión 4.9.1 pero corregí el/los valor/es y funciona bien

Ghostkeeper, 2021-05-20:
Could you communicate in English? We don't speak every language in the world, and English is the norm here.

We still need a project file to be able to debug this error. The Regular/Maximum Fan Speed Threshold seems to be working fine as far as I can tell, and so does the Minimum Layer Time setting.

closed 2021-05-28 by no-response[bot]