Ultimaker/Cura #9815: Modify settings for overlaps not as user friendly as is could be

nickbuhling, 2021-05-12:
### Is your feature request related to a problem?

Dear developers,
I am not sure whether this is intended or it is a bit of a bug, but when you use "modify settings for overlaps" it sort of duplicates the settings and applies them twice instead of modifying the selected settings. So you get 2 times 3 walls for example or dubble expanded skin on top of each other (leading to print issues). Also, it will make a bottom layer inside the print which requires you to select bottom layers and set it to 0 as well. 

### Describe the solution you'd like

I think it would be more user friendly if all the settings stay the same and only the selected settings are changed. For example only changing the infill density and leaving the wall count and all other things the same. 

### Describe alternatives you've considered

Setting all the settings manually

### Affected users and/or printers

Everyone

fvrmr, 2021-05-14:
Hi @nickbuhling thank you for your feature request.
Do you mean the cutting mesh or the infill mesh? The settings for infill mesh are always on 0. 
Here you can find a similar issue: https://github.com/Ultimaker/Cura/issues/5652

nickbuhling, 2021-05-14:
Hi,
I indeed mean the infill mesh. Unfortunately the settings are not all on 0. For example when I include the bottom layers setting is shows to be set to (in this example) 7. Same for the top layers. Do you know why this is?

Thanks for your reply!

fvrmr, 2021-05-17:
Hi @nickbuhling could you share a project file?
When I include the bottom layers, it is set to 0.
[image]

nickbuhling, 2021-07-06:
Sorry for my late reply. Here is a screenshot of what is happening when I do "modify settings for overlap". The rest I would have to manually add and set them to zero. For example also the top and bottom amount of layers. Hope this helps.

Here is the project file:
[Issue_example.zip](https://github.com/Ultimaker/Cura/files/6769604/Issue_example.zip)

[image]

Ghostkeeper, 2021-07-14:
The problem here is really in the distinction between wall thickness and wall line count. And similarly, top/bottom thickness and top/bottom layers.

The infill mesh is expected to default to having no shell. This is its most common usage, especially among novices (experts know that increasing the number of walls is often more effective to increase strength).

The two settings Wall Thickness and Wall Line Count effectively control the same thing, but are different units. The way it's implemented lets the Wall Line Count be the setting that is actually used for slicing, and that setting normally gets automatically calculated depending on the Wall Thickness. The intention is that Wall Thickness is the setting most people will want to adjust, since it's more intuitive for a layperson who doesn't go into the specifics of FFF printing.

But in this case the problem is that the user overrides the Wall Line Count setting before creating the infill mesh. The line count then no longer depends on the wall thickness. When switching to an infill mesh, the Wall Line Count setting is copied from the extruder settings. But instead of that copying the formula that automatically adjusts the number of lines based on the thickness, it copies the fixed wall line count. The thickness is set to 0 as per the default, but the Wall Line Count remains a fixed value, 2 in your case.
With the Top/Bottom Thickness it doesn't even copy the value but just lets it inherit the Top Thickness and Bottom Thickness; it's not overridden as a per-object setting and thus will be the extruder's value, which fixes the top and bottom layer count to 6.

I'd classify this as a bug. The expected behaviour is that adding an infill mesh doesn't add a shell by default. That's not what's happening here. It's logical that it doesn't happen based on how it's implemented, but that is not what the user expects to see.

As a solution, I think we'll need to override the Wall Thickness and Top/Bottom Thickness to 0 as we do now, but also override the Wall Line Count and Top/Bottom Layers to *the formula that makes it inherit from the thickness setting* rather than hard-coding it to 0 or making it copy from the extruder settings.

Ghostkeeper, 2021-07-15:
Developers, see issue CURA-8393.

commit 6f6d8d0f (2021-07-15): Override all settings that might influence infill mesh shells to 0

If the user had specialised settings fixing the number of walls or skin layers to an integer number, this fixed value would get copied over from the extruder settings, causing infill meshes to have walls by default.
This is unintuitive. We want infill meshes to behave like it's only infill by default. So remove all of those overrides so that only infill gets printed there, by default.

The user may still change the number of walls or skin layers or their parent settings. But by default it should now behave like infill within i [... cut]
closed 2021-07-15 by Ghostkeeper