Ultimaker/Cura #9816: Cura 4.9 Linux Market place No Longer works

TomTay56, 2021-05-12:
### Application Version

4.9.0

### Platform

Linux Mint 20.1

### Printer

Ender-3

### Reproduction steps

Clicked on Market Place.

### Actual results

Market Place window opened up. 
I even created an account and logged in for this.
[image]

[cura.log](https://github.com/Ultimaker/Cura/files/6467916/cura.log)

### Expected results

It used to display materials and plugins.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I even tried deleting all things cura and ultimaker. 
Tried an older version app image.
And still have the problem.
I update Linux weekly so they might have changed something to make this happen.

fvrmr, 2021-05-14:
Hi @TomTay56 thank you for your report.
This week we released 4.9.1: https://github.com/Ultimaker/Cura/releases/tag/4.9.1
Could you check if it also happens there?

TomTay56, 2021-05-17:
4.9.1 still no market place.
Also try to open configuration folder get archive not supported.
I deleted all configuration and everything still the same.

[image]

Ghostkeeper, 2021-05-20:
I'm not sure what you were trying to do with that last screenshot. Archive type not supported? There is an archive called "4.9"? And it's a file format that your computer doesn't support?

To the point, I think this is the relevant part of the log file:

`2021-05-12 13:04:38,268 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[d9a8ad2f][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError SSL handshake failed. The server returned: b''`

So it doesn't make a secure connection to the ultimaker.com website to request the packages from the Marketplace API. This is similar to the browser giving you this error upon visiting websites with broken SSL: https://wrong.host.badssl.com/

When I visit the Marketplace (also using Linux) it works fine. So something is different about your computer that causes this. There are a couple of things I can think of:

* Your system clock is off course. Since the SSL handshake uses timestamps to validate responses, a reasonably accurate system clock is important.
* Someone or something is intercepting your traffic. This is what SSL is meant to prevent: Nobody can snoop on the traffic nor change anything in the request or the response. If something is trying to make a change to the traffic (e.g. Chinese Firewall or intrusive employee surveillance software), it would give this error. In a free country, it's more likely that it's something you installed unknowingly, like a less than savoury VPN or a virus.
* You've installed a custom certificate authority which is not sound. This could come with a VPN, but also with company surveillance programs like FortiNet.
* The connection is spotty. Perhaps it's just not able to contact the certificate authorities or something.

So far we've been unable to reproduce this problem. It seems to work fine (and we're seeing lots of people successfully downloading these plug-ins and materials on all OSes).

TomTay56, 2021-05-21:
So being I also found out the Ghostkeeper made comments about Macs having the same problem. I reinstalled linux after 10 years. Took awhile. Could not figure out what actually was wrong with nss and ssl. Now that I have reinstalled Linux Mint 20.1. It works fine.

TomTay56, 2021-05-21:
Works Now.

closed 2021-05-21 by TomTay56