Ultimaker/Cura #9817: Instant shutdown

DKNicholson, 2021-05-12:
### Application Version

4.9.1

### Platform

Linux Mint Cinnamon

### Printer

Anycubic Mega S

### Reproduction steps

1. Imported a project as an .STL file
2. Able to print it without any alteration
3. Needed to re-scale object

### Actual results

As soon as I selected the object and clicked on the 'Scaling' icon on the left side, Ultimaker Cura instantly shuts down.

### Expected results

I should have been able to scale the object

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6467989/cura.log)

This is the latest logfile - hopefully it will highlight the issue, but I have since used 4.8.0 in order the get the job done.

smartavionics, 2021-05-13:
For an alternative Cura release that has this problem fixed, see https://github.com/smartavionics/Cura/releases/tag/20210511

fieldOfView, 2021-05-13:
Duplicate of #9709

fvrmr, 2021-05-14:
Closing since it is a duplicate of #9709

closed 2021-05-14 by fvrmr