Ultimaker/Cura #9818: .stl icons not showing on Windows 10 Desktop

jimmysdiy, 2021-05-12:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

I am using Windows Surface Laptop 2.

I uninstalled the 4.9.0 after I received the notification to update to 4.9.1. After installing 4.9.1, Cura registers itself as a default app but the icons don't show Cura's icon.

### Actual results

The icons don't register or shows up properly

I uninstalled and reinstalled once again, and no change

I rebuilt the icon cash and no change

### Expected results

The Icons should show cura's icon logo

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fieldOfView, 2021-05-13:
Have you tried logging out of Windows and back in again?

jimmysdiy, 2021-05-13:
So I tried following
reboot the computer - the issue is not fixed
signed out of profile and logged back in - the issue is not fixed
Cleared Icon Cache - the issue is not fixed

Uninstall and Reinstalled Cura 4.9.1 - Not fixed
Uninstalled Cura 4.9.1 and Installed Cura 4.9.0 - Issue fixes
[image]
Then Uninstall Cura 4.9.0 and Install 4.9.1 - the issue comes back and seems like the issue is with .stl extension

jimmysdiy, 2021-05-13:
I was able to fix it. Below is how I fixed

I opened regedit
went to Computer\HKEY_CLASSES_ROOT\.gcode.4.9\DefaultIcon
Clicked on Default and remove the path.

Then I rebooted and the icons came back.

fvrmr, 2021-05-14:
Hi @jimmysdiy thank you for your bug report and the work around.
I will discuss this issue with the team.

yet-another-average-joe, 2021-05-14:
Similat problem here : not with stl, but with 3mf : no more icon.

jimmysdiy, 2021-05-14:
You can follow same steps as I did. See if that fixes it. Then post back for the team

yet-another-average-joe, 2021-05-15:
I took time to tinker with the registry.
I deleted all hives : .3MF, .3mf4.6, .3mf4.8, .3mf4.9, then right clicked "open with" a 3mf file, and chose Cura.exe as default app.
The 3mf icon is back. A new "HKEY_CLASSES_ROOT\.3mf was created, its contents is : default = ".3mf.4.6"
Didn't have to reboot nor to unlog/log.

Cura.exe appears under :
\HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\FileExts\.3mf\UserChoice

fvrmr, 2021-05-17:
We have created a ticket on our backlog for this.
Developers see CURA-8239

fvrmr, 2022-05-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]