Ultimaker/Cura #9820: When sliced there is a gap which is not in the pre-sliced model

brinuk, 2021-05-13:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Loaded a model (STL file) to slice. Sliced the model

[image]
[image]

### Actual results

There is an horizontal gap in the sliced model

### Expected results

The sliced model should be the same as the pre-sliced model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_camera support Iflight XL10 v2.zip](https://github.com/Ultimaker/Cura/files/6471569/CE3_camera.support.Iflight.XL10.v2.zip)

GregValiant, 2021-05-13:
Even after being repaired (the model isn't watertight) the problem remains and it's in the model.  If you turn on "Print Thin Walls" you will see that the wall thickness almost disappears there.  That area isn't connected well to the other features either.  Switching to a .2 nozzle (just as a test) helps to see the problem.

fvrmr, 2021-05-14:
Hi @brinuk thank you for your report.
This has to do with your model.
Breaking it up and checking the width of the wall. You can see that the walls are to thin to print with the wall width. That's why changing the nozzle like @GregValiant would work. Or you could change the models walls. 

[image]

Let me know if this makes sense to you.

brinuk, 2021-05-15:
Thank you for your prompt response. I can see the cause of the problem now. I have redesigned the model and it now prints well. Thank you.

closed 2021-05-15 by brinuk