Ultimaker/Cura #9822: Back Slash in saving Gcode

WForret, 2021-05-13:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Crality Ender 3 Pro

### Reproduction steps

Installed the extensions : Gcode Filename Format
Wanted to save my program, but didn't worked because of the / tat the beginning

### Actual results

Didn't save my project

### Expected results

To delete the back slash befre saving the projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-05-13:
Please log an issue with the creators of that plugin instead:
https://github.com/rgomezjnr/GcodeFilenameFormat/issues

fvrmr, 2021-05-14:
I will close this issue since it is not relevant to Cura but to the plugin.

closed 2021-05-14 by fvrmr