Ultimaker/Cura #9825: Cura freezes on offline startup

thekojent, 2021-05-13:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

1. Turned off network
2. Opened Cura

### Actual results

Cura goes through startup sequence and freezes on "Initializing Build Volume" stage

### Expected results

Cura finishes startup sequence and opens successfully.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6474907/cura.log)

fvrmr, 2021-05-14:
Hi @thekojent thank you for your bug report.
I tried to reproduce this, but I couldn't.

I will discuss this with the team. Keep you posted.

Ghostkeeper, 2021-05-21:
We've been getting a bunch of these reports and even more crash reports via the crash reporting tool. We're still having trouble reproducing the crash ourselves, but we're investigating!

In the end, this duplicate is winning out as the one with the most information: https://github.com/Ultimaker/Cura/issues/9847 Please continue there!

closed 2021-05-21 by Ghostkeeper