Ultimaker/Cura #9826: 4.9.1 Calculated run time incorrect

ipter, 2021-05-13:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

BIQU B1

### Reproduction steps

Sliced a stl that I had done in v4.9.0 and the runtime said 6.5 hrs.
I loaded the gcode into octiprint and it calculated the run time to be 2 hrs.

### Actual results

Slices fine, runs fine bad estimated runtime

### Expected results

Better estimated runtime

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-05-14:
Hi @ipter thank you for your report.
The time estimation is based on Marlin from 5 years ago and on the printer settings in the def.json file. 
And we don't maintain these settings for the BIQU B1 printer. 

Let me know if that makes sense to you.

minhncc, 2021-05-15:
Same issue with Creality Ender3, sliced on 4.9.0 is 1:45 hours, on 4.9.1 is 2:23 hours. 
Actual printing took 1:47 hours.

Theguffy1990, 2021-05-15:
> Hi @ipter thank you for your report.
> The time estimation is based on Marlin from 5 years ago and on the printer settings in the def.json file.
> And we don't maintain these settings for the BIQU B1 printer.
> 
> Let me know if that makes sense to you.

Persists with Ender 3 V2. 

Sliced file: 37 hours, 23 minutes.
Actual print time: 25 hours, 29 minutes, 50 seconds.

If it were around 10% off, i wouldn't mind, but over 30% off? I think that's a bit much, and should probably be investigated further.

Ghostkeeper, 2021-05-21:
We have verified that the time estimation is correct for the firmware of Ultimaker printers (which run on Marlin) aside from a slight variation caused by the clock speeds of different chips. The implementation of time estimation is also based on Marlin, effectively running the same algorithms to determine the speed, acceleration and jerks, except for that the firmware rounds it to steps or microsteps while Cura assumes infinite precision (or almost: double floating point precision). We've experimentally tested that on all of our printers the time estimation is always within 5% even if heating up and cooling down speeds are measured along (which aren't estimated by Cura).

Of course that DOES depend on all of the parameters being correct in Cura. The time estimation depends on the following parameters being in sync with your firmware:
* Maximum Speed X/Y/Z
* Maximum Feedrate
* Maximum Acceleration X/Y/Z
* Maximum Filament Acceleration
* Default Acceleration
* Default X-Y Jerk
* Default Z Jerk
* Default Filament Jerk
* Minimum Feedrate

What values are you using for those settings in Cura and in your firmware? And are you running stock firmware?

minhncc, 2021-05-21:
I uninstalled 4.9.1 and reinstall 4.9.0 to resolve the issue. There must be something change between them that made the huge incorrect. 
Maybe never update if the estimation incorrect from now on.

Ghostkeeper, 2021-05-21:
The only thing that changed between those two versions is [a fix for priming when using the G280 prime blob](https://github.com/Ultimaker/CuraEngine/compare/4.9.0...4.9.1) which isn't supported by the Ender printers (and has nothing to do with time estimation either). I think something else might be different about those tests. Perhaps you were using a different print or different settings? Again:

> What values are you using for those settings in Cura and in your firmware? And are you running stock firmware?

minhncc, 2021-05-21:
I use all defaut settings between two Cura versions, the same Ender 3 with stock firmware.
Only change is the estimation time.

Ghostkeeper, 2021-05-31:
The time estimation algorithm [hasn't changed in 2 years](https://github.com/Ultimaker/CuraEngine/blob/master/src/timeEstimate.cpp) so I think something else must've changed that you've been overlooking.