Ultimaker/Cura #9827: Cura 4.9.1 slow for every click / not operational

brunoosti, 2021-05-14:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Anycubic i3 mega s

### Reproduction steps

1. open cura 4.9.1
2. try to click or do any operation, cura requests from processor and takes several seconds to execute. 

### Actual results

Cura freezes for a while and than execute the task. Insisting while loading leads it to crashing the software.

### Expected results

Not able to operate Cura because of slow response. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is not happening on 4.8 and Arachne beta

fvrmr, 2021-05-14:
Hi @brunoosti thank you for your report.
Could you share your log file?
You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

brunoosti, 2021-05-14:
Sorry, I didn't know where to find it. There you go (attached to the first thread message).

fieldOfView, 2021-05-14:
Could you try disabling the SidebarGUI plugin? There seems to be an issue with my plugin that I have sofar been unable to reproduce. If disabling it helps, please close this issue and see https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/24

brunoosti, 2021-05-14:
It did! Thanks! 
Something in 4.9 is making it crash..  
Arachne is fine.

closed 2021-05-14 by brunoosti

fieldOfView, 2021-06-23:
For the benefit of other people stumbling upon this thread, disabling the FancyZones PowerToy seems to fix this problem.