Ultimaker/Cura #9831: Right mouse button

metricastudio, 2021-05-14:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3v2

### Reproduction steps

right mouse button works for rotating viewport but also is opening a tools popup 
you cannot rotate model without that annoying popup 

### Actual results

cannot rotate viewport without popping a tools menu

### Expected results

i just want it to rotate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no

fvrmr, 2021-05-17:
Hi @metricastudio thank you for your report.
Does the popup happen when you have the right mouse button pressed? 
Because it is correct that when you click the menu will show up.

closed 2021-05-31 by no-response[bot]

Ghostkeeper, 2021-05-31:
The context menu is only supposed to pop up if the camera didn't rotate (which has a jitter leniency of 1% of the window size). If it does pop up even when the camera is moved, you might have a mouse that has some bounce in the buttons. That's pretty common with mice that are getting old, as dirt gets into the switch.