Ultimaker/Cura #9832: Build Plate initial layer temperature not set because no variable exists

BaileyMillerSSI, 2021-05-14:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Snapmaker A350

### Reproduction steps

1. Changed initial build plate temperature to 10 degrees warmer
2. Changed start gcode variable to material_bed_temperature_initial_layer
3. Sliced and evaluated GCODE
4. Variable not replaced

### Actual results

The M140 command is set to the build plate temperature everywhere, no mention of the initial build plate temperature even once in the GCODE. Variable isn't replaced with initial build plate temperature.

### Expected results

M140 and M190 commands are set to the initial build plate temperature.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

[Projectfile.zip](https://github.com/Ultimaker/Cura/files/6480364/Projectfile.zip)

GregValiant, 2021-05-14:
{material_bed_temperature_layer_0} is the keyword you want.  The keyword list is [HERE](http://files.fieldofview.com/cura/Replacement_Patterns.html)

fvrmr, 2021-05-17:
Hi @BaileyMillerSSI thank you for your report.
I will close this issue. Since @GregValiant suggestion would solve your problem.
But feel free to reach out if you have more questions.

closed 2021-05-17 by fvrmr