Ultimaker/Cura #9834: Beginning of layer on straigth line instead of corner

matthieut59, 2021-05-15:
### Is your feature request related to a problem?

Corners are often the weak spot on my prints (warping etc.), and on some simple prints (mostly rectangle) I would find it very usefull to start my first layer for instead on the inner wall of a straigth line, instead of the corner.

Not matter what I m trying to define in the slice, it always starts on a corner :( Random is clearly not random and always pick a corner :( )

Thks in advance, interested by your view on this!

Matt

### Describe the solution you'd like

anything not limiting the seam position to a corner would be usefull

### Describe alternatives you've considered

none so far :(

### Affected users and/or printers

All users

smartavionics, 2021-05-15:
My Cura will start walls not on a corner when the z-seam alignment is user specified and the seam corner preference is set to none. Like this...

[image]

My latest release can be found at https://github.com/smartavionics/Cura/releases

Please read the README.md there for more information.

matthieut59, 2021-05-15:
I tried pretty much all possible Z seam X/Y couple, and it s always on a corner with this setting :  
[image]

smartavionics, 2021-05-15:
Which Cura are you using? This only works with my releases, not the UM releases.

matthieut59, 2021-05-15:
oh! I m using master 4.9, so I need to move on your version ? Have you proposer this enhancement to Master ?

smartavionics, 2021-05-15:
No, sorry, I don't contribute new features to Cura any more, just bug fixes.

Ghostkeeper, 2021-05-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/8010.

closed 2021-05-17 by Ghostkeeper